arduino/Arduino #3188: Problems with more or equal than 3 of char !

josewilker, 2015-05-20:
When you put more or equal than 3 of (!) symbol on a Serial.println ... like, [Serial.println("MATCH!!!!");]... compiler it's ok ... but when I try upload sketch to arduino, show problems with AVR and a lot of timeout.

Try:

Serial.println("MATCH"); - WORKS
Serial.println("MATCH!"); - WORKS
Serial.println("MATCH!!"); - WORKS
Serial.println("MATCH!!!");  - PROBLEM : avrdude: stk500v2_ReceiveMessage(): timeout

PaulStoffregen, 2015-05-20:
Maybe you've using an ancient Arduino Mega?  Or maybe a Chinese clone, programmed with the very old bootloader?

ffissore, 2015-05-20:
Indeed. It looks like a problem solved in #1183. Please upgrade bootloader. Reopen if upgrading it doesn't fix

closed 2015-05-20 by ffissore

josewilker, 2015-05-26:
uow! tks! \o/

PaulStoffregen, 2015-05-26:
Assuming this problem was due to a clone or counterfeit still using the ancient bootloader, you might try posting negative feedback on whatever site sold the board, or elsewhere, to warn other users that they're still shipping the buggy bootloader.