arduino/Arduino #3192: Non-persistent formatting in build/shared/lib/preferences.txt

Arnie97, 2015-05-20:
Some entries have leading and trailing whitespace around equality signs, while others not.
There are even some tabulators on line 127.

ffissore, 2015-05-21:
Can you provide a fix with a pull request?

Chris--A, 2015-05-23:
@Arnie97 
Have a look at #3205, does this contain the changes you are after?

Arnie97, 2015-05-23:
@Chris--A 
Nice job. Thanks.
Should we remove the spaces before end of each line?

Chris--A, 2015-05-24:
I do not think it is necessary. The changes I made were purely for some eye candy. The spaces at the end of the file aren't visible unless highlighted/selected.

However if you would like to make the fix, feel free to add a commit to my branch and I'll merge it in then it'll appear in the PR (#3205).
https://github.com/Chris--A/Arduino/tree/tidy-preferences

closed 2015-05-25 by Arnie97