arduino/Arduino #3195: Incorrect arm-none-eabi-g++ path 1.6.4+ OSX

BrentWilkins, 2015-05-20:
I've enabled 'Arduino SAM Boards (32-bits ARM COrtex-M3) by Arduino version 1.6.4' in the 'Boards Manager'.  I cannot compile simple sketches such as Blink.  Here is the IDE output:

/Users/brent/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10605 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -I/Users/brent/Library/Arduino15/packages/arduino/hardware/sam/1.6.4/system/libsam -I/Users/brent/Library/Arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/CMSIS/Include/ -I/Users/brent/Library/Arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/Device/ATMEL/ -I/Users/brent/Library/Arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino -I/Users/brent/Library/Arduino15/packages/arduino/hardware/sam/1.6.4/variants/arduino_due_x /var/folders/9k/f64_hhq917l4cs5t_cplxf0r0000gn/T/build5252173903680666303.tmp/Blink.cpp -o /var/folders/9k/f64_hhq917l4cs5t_cplxf0r0000gn/T/build5252173903680666303.tmp/Blink.cpp.o 
**Cannot run program "/Users/brent/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" (in directory "."): error=2, No such file or directory**

Sure enough, that doesn't exist.  These two directories exist:
/Users/brent/Library/Arduino15/packages/arduino/tools/(arm-none-eabi-gcc|avr-gcc)

The avr-gcc dir contains binaries, but the arm-none-eabi-gcc only contains a single lib.  I can find the compiler in the install directory:
/Applications/Arduino1.6.1.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/arm-none-eabi-gcc

I'm running 1.6.4 and this mornings hourly, so I don't know the path if I didn't ever install that version.

ffissore, 2015-05-20:
Which board have you selected in boards menu?

BrentWilkins, 2015-05-20:
Functions the same with either port of the Arduino Due.

ffissore, 2015-05-25:
With #3141 we fixed an error which may lead to your error message. Can you please:
* delete `~/Library/Arduino15/packages/arduino/hardware/sam/`
* delete `~/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/`
* delete `~/Library/Arduino15/packages/arduino/tools/bossac/`
Reinstall SAM support and retry?

ffissore, 2015-06-04:
Given the lack of feedback, I suppose the issue can be closed. Reopen if hourly build still shows it

closed 2015-06-04 by ffissore

BrentWilkins, 2015-06-04:
Sorry I missed this (and likely others) among the hundreds of emails from
this list.  I tested it and it seems to be fixed.  thanks.