arduino/Arduino #3196: Suggestion: make MIN_PULSE_WIDTH and MAX_PULSE_WIDTH editable by user

bpinhosilva, 2015-05-20:
Recently, I discovered that my servo motor works a little bit different in terms of the pulse I should send to it in order to move from 0 to 180 degrees. I did a test by changing the pulse width and noticed that it works from 600 to 2100 (min and max respectively). 

I was not able to use it with Arduino's Servo library, but when I went through the source code I saw these lines:

#define MIN_PULSE_WIDTH       544     // the shortest pulse sent to a servo  
#define MAX_PULSE_WIDTH      2400     // the longest pulse sent to a servo 

Then I changed it to my values and now my servo works good.

I would like to make a suggestion to include two member functions: setMinPulseWidth() and setMaxPulseWidth() so the user could address the servos properly.

This way, someone could use writeMicroseconds() to detect the correct interval for max and min angles and set them by the functions listed above.

fiechr, 2015-05-26:
Please correct me if I'm wrong or misunderstood you, but isn't that already possible?
```
servo.attach(pin, min, max) 
```
> min (optional): the pulse width, in microseconds, corresponding to the minimum (0-degree) angle on the servo (defaults to 544)

> max (optional): the pulse width, in microseconds, corresponding to the maximum (180-degree) angle on the servo (defaults to 2400) 

http://www.arduino.cc/en/Reference/ServoAttach

So, in your case:
```
servo.attach(PIN_SERVO, 600, 2100);
```

bpinhosilva, 2015-05-26:
Yes, that's right! I checked the new version and it is working properly. Thanks

closed 2015-05-26 by bpinhosilva