arduino/Arduino #3198: Selected Board preview doesn't match the board selected in the dropdown

BrentWilkins, 2015-05-20:
I was working on one project on a Pro Mini, but opened a new window with the example Blink sketch.  I uploaded that to another board & closed that window.  The parent window is now in a bad state.
[image]

ffissore, 2015-05-21:
I'm unable to reproduce it. Here are the steps I followed:
1) Opened IDE
2) Selected Arduino UNO
3) New sketch window
4) Selected Arduino Leonardo
5) Compiled empty sketch
6) Closed window
7) Now first window correctly displays Arduino Leonardo

Are these the steps you've taken?

ffissore, 2015-05-21:
Also tested on mac and still unable to reproduce it.

BrentWilkins, 2015-05-22:
I'm also having trouble reproducing it.  Glad I took a screenshot.  I'll report back later if I see it again.  It's apparently non-trivial to get it to that state.

closed 2015-05-22 by BrentWilkins