arduino/Arduino #3210: MacOSX: don't switch to the dedicated graphic card

ffissore, 2015-05-22:
It looks like the IDE is triggering the more powerful GPU of the two some Macs have installed.
AppBundler supports a flag that disables such switch
https://bitbucket.org/infinitekind/appbundler/pull-request/6/add-support-for/diff

ffissore, 2015-06-03:
#3225 closed because of lack of feedback. Apparently, too few users are affected by the issue

kriegsman, 2015-08-05:
Actually @ffissore, I think that basically ALL Mac laptop users are affected by this issue, but since 'graphics card switching' is automatic and invisible under OSX, 99% of users can't tell WHY running the Arduino IDE is draining their battery so quickly.  (Installing "gfxCardStatus" makes it abundantly clear where the power is going.)

I'd like to suggest that we re-merge the fix for this. If you'd like more feedback, I will happily round up a dozen Mac laptop Arduino users and get them to report what they're finding.

Please let me know what would help get the patch re-merged and tested for acceptance.

Thanks -Mark.

kriegsman, 2015-08-05:
Actually, using the built-in Activity Monitory, you can see which applications are demanding use of the 'high performance GPU' (which is what also draws much more power).  Attached are a couple of screenshots, showing the Arduino IDE sitting 'idle' but still using significant power.  Also worth noting that many MacBooks have these dual graphics chips: one energy-efficient one, and one high-performance one.  I believe they introduced this hardware first in 2008, and it's still present in the the lastest higher-end MacBook Pros.

[image]

[image]

I'd love to help out all the Arduino developers who are using MacBooks improve their battery life, and let them hack longer between charges.

Thank you so much for all your hard work.  It's hard to express how fantastic the whole Arduino project and ecosystem is.

ffissore, 2015-08-05:
Happy to see you back.
First two test I need you to perform:
1) download latest [hourly](http://www.arduino.cc/en/Main/Software#hourly), edit file `Arduino.app/Contents/Info.plist`: locate row
```
<string>-Xmx512M</string>
```
and, below, add row 
```
<string>-Dsun.java2d.opengl=false</string>
```

Does you mac still flag the IDE as requiring a high perf GPU?

2) Download latest [Processing](https://processing.org/download/) and start it. Does you mac flag it as requiring a high perf GPU?

kriegsman, 2015-08-06:
Even with the "-Dsun.java2d.opengl=false" in the Info.plist, the Arduino IDE app is flagged as requiring the High Performance GPU, as reported by both Activity Monitor and gfxCardStatus.

The latest Processing (v2.2.1) is also flagged as requiring the High Perf GPU.

kriegsman, 2015-08-06:
Unfortunately, this ticket may be relevant: "JDK-8041900 [macosx] Java forces the use of discrete GPU" https://bugs.openjdk.java.net/browse/JDK-8041900
My MacBook that I've been testing on is running java version "1.8.0_51".

Also, I do note that "NSSupportsAutomaticGraphicsSwitching" is present in the Info.plist with a value of true in both the hourly build as well as in 1.6.5.   It's looking like we'll need at least the NSSupportsAutomaticGraphicsSwitching AND the java2d.opengl=false ... AND something else, it seems.

ffissore, 2015-08-07:
Pretty recent too. I guess we'll just sit and wait for a fix from upstream.

kriegsman, 2015-08-07:
Thank you for the time and attention, and all your work. When they fix it upstream, we'll hopefully get the right behavior "for free".

ArekSredzki, 2015-08-07:
Looks like it is an upstream issue, but I'd like to offer my support in fixing this issue if it persists after the upstream problem has been fixed. This has been a huge problem for all too long.

kriegsman, 2015-12-07:
It now looks like the upstream issue has been addressed.
https://bugs.openjdk.java.net/browse/JDK-8041900
Thoughts on "what next" now?  I'm (still) very happy to help test any revisions and give prompt feedback.

bafonso, 2015-12-07:
I'd be happy to test this too, it is a big resoure sink to switch to the discrete gpu.

ffissore, 2015-12-07:
thank you @kriegsman it seems they've really fixed it. If I get the labels right, fix is in 8u76 which hasn't been released yet. Latest stable jvm is [8u66](http://www.oracle.com/technetwork/java/javase/downloads/index.html), latest beta is [8u72](https://jdk8.java.net/download.html). I guess we just have to sit and wait for the fix to be released

kriegsman, 2015-12-07:
Thanks for the cross-check, and I think I agree with your reading of the labels. I will now resume waiting.

kriegsman, 2016-04-08:
The current stable jvm from Oracle appears to now be 8u77, which if I read the labels correctly, includes a fix for the GPU-switching issue.  I've installed 8u77 as my system jvm, and I just downloaded the latest hourly Arduino IDE build -- and I'm still seeing the switch to the high-power GPU when I launch the Arduino IDE. 
Is there other information I could provide, or tests I could run, that would be helpful?

facchinm, 2016-04-08:
The IDE bundles its own jvm, so the one installed in the system has no effect. If you could provide a link to the changelog reporting  that the issue is closed we can update the bundled jvm and have it available in nightly builds!

kriegsman, 2016-04-08:
Thanks very much @facchinm ... I'm trying to track it down, and I apologize but I'm now I'm a bit confused about which jvm version it is or isn't fixed in.  This seems to be the relevant ticket https://bugs.openjdk.java.net/browse/JDK-8041900.  Now that I'm reading it again, maybe it's saying that it's fixed in 8u92?  If that's the case, then the fix may only be available in the jvm betas, which wouldn't help us here.

facchinm, 2016-04-08:
Thanks a lot @kriegsman for the links! It really looks like we need to wait for u92 to be released to inherit the fix...

keilunsoftware, 2016-06-18:
Just installed the u92 version. Problem still present...

badgeek, 2016-07-20:
+1 @keilunsoftware

kptdobe, 2016-08-24:
+1, issue still valid

kptdobe, 2016-08-24:
but seems to be fixed in 1.6.10: I do not get the "Requires High Perf Graphic" flag anymore in the Activity Monitor.

kriegsman, 2016-08-24:
Just checked Arduino 1.6.11 on OSX 10.11.6 -- good result: no more "Requires High Perf GPU".  Also "Energy Impact" of Arduino hovering near zero when IDE is idle, further evidence that this is better now.

facchinm, 2016-08-26:
Closing as fixed

closed 2016-08-26 by facchinm

bafonso, 2016-08-29:
Does anyone know what changed to make this be fixed?

facchinm, 2016-08-30:
We are now bundling JRE update u92 which contains the fix :smile:

commit 45b61ef0 (2015-05-22): MacOSX: set NSSupportsAutomaticGraphicsSwitching to true, so the OSX can switch back to the integrated GPU and save battery. Fixes #3210
commit 6c2a6aef (2015-05-25): Revert "MacOSX: set NSSupportsAutomaticGraphicsSwitching to true, so the OSX can switch back to the integrated GPU and save battery. Fixes #3210"

This reverts commit 45b61ef02bd139180a745b642385789a8cd69ce2.