arduino/Arduino #3212: Remove deprecated ModbusMaster Library

4-20ma, 2015-05-22:
Thank you for adding my `ModbusMaster` library to the list.

To be more consistent with other libraries, I removed the word ` Library` from the `name` field of `library.properties` and now it's showing up twice in the Library Manager. Please remove the extraneous ones (there are 2).

Now that I understand how it shows up, I'll keep this field consistent.

http://downloads.arduino.cc/libraries/library_index.json

```
    {
      "name": "ModbusMaster Library",
      "version": "0.10.0",
      "author": "4-20ma",
      "maintainer": "Doc Walker \u003c4-20ma@wvfans.net\u003e",
      "sentence": "ModbusMaster libary",
      "paragraph": "This is an Arduino class library for communicating with Modbus slaves over RS232/485 (via RTU protocol).",
      "website": "https://github.com/4-20ma/ModbusMaster",
      "category": "Communication",
      "architectures": [
        "avr",
        "sam"
      ],
      "types": [
        "Contributed"
      ],
      "url": "https://github.com/4-20ma/ModbusMaster/archive/v0.10.0.zip",
      "archiveFileName": "ModbusMaster_Library-0.10.0-github.zip",
      "size": 365030,
      "checksum": "SHA-256:9e042373d67ea155762e7cb6c47eb39e53f2f6a6b7d5b725aad6065e93d6d920"
    },
    {
      "name": "ModbusMaster Library",
      "version": "0.10.1",
      "author": "4-20ma",
      "maintainer": "Doc Walker \u003c4-20ma@wvfans.net\u003e",
      "sentence": "ModbusMaster libary",
      "paragraph": "This is an Arduino class library for communicating with Modbus slaves over RS232/485 (via RTU protocol).",
      "website": "https://github.com/4-20ma/ModbusMaster",
      "category": "Communication",
      "architectures": [
        "avr",
        "sam"
      ],
      "types": [
        "Contributed"
      ],
      "url": "https://github.com/4-20ma/ModbusMaster/archive/v0.10.1.zip",
      "archiveFileName": "ModbusMaster_Library-0.10.1-github.zip",
      "size": 674951,
      "checksum": "SHA-256:fea5b5b1781ac52226f73544adb577557ecc38e23ba77c43be52847eaeda9f0f"
    },
```

ffissore, 2015-06-03:
Have you removed the tags showing the wrong name? Until then, the wrong/double name will keep on showing up

ffissore, 2015-06-03:
FYI I've added an entry in the FAQ https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#sorry-i-did-something-wrong-how-can-i-change-or-unpublish-an-already-published-library

ffissore, 2015-06-04:
Looks like you did remove those tags. Closing

closed 2015-06-04 by ffissore

4-20ma, 2015-06-05:
@ffissore yes, I did remove the tags; I was hoping this would make the library releases disappear from the json download, but somehow they remain in the list.

ffissore, 2015-06-05:
As a rule of thumb, you should not remove tags and we don't expect you to remove them. Tags are library versions from a user PoV. Removing a tag (and its version) will make the IDE unable to recognize an installed library as one coming from the Library Manager, causing headaches to users. So, whenever possible, don't remove tags