arduino/Arduino #3213: Sugestion: unused variables notification

anjoze, 2015-05-22:
When compiling, some IDEs warn about variables not used.
Its a nice feature especially for sketches with tons of variables.
Thanks

per1234, 2015-05-24:
In IDE 1.6.4+ If you set **File > Preferences > Compiler warnings** to **More** or **All** then you will get warnings about unused local variables. Unfortunately no warnings about unused global variables.

cmaglie, 2015-05-25:
The global variable is not detected because it may be used from another compilation unit. If you declare the global variable as `static`, it can not be used anywhere else and it's correctly detected as unused, in fact the following sketch:

```C++
int a;
static int b;
void setup() {
  int c;
}
void loop() { }
```

produces:

```
sketch_may25a.ino: In function 'void setup()':
sketch_may25a.ino:5:7: warning: unused variable 'c' [-Wunused-variable]
sketch_may25a.ino: At global scope:
sketch_may25a.ino:2:12: warning: 'b' defined but not used [-Wunused-variable]
```

closed 2015-05-25 by cmaglie

per1234, 2015-05-25:
Thanks for the information cmaglie!