arduino/Arduino #3215: "Invalid version found:" with library name

mischnic, 2015-05-24:
After "upgrading" several library to include a corresponding library.properties file and instally Teensyduino which installs builtin librarys, I get many very unhelpful errors:  
Maybe include a string of the library/platform name to ```VersionHelper.valueof``` to make this error more informative? This is especially hard to do in  VersionComparator.java as the Java Comparator doesn't allow something like that. But it is otherwise impossible to track down what causes these errors.

```
Invalid version found: 
Invalid version found: 
Invalid version found: 
Invalid version found: 
Invalid version found: 
Invalid version found: 
Invalid version found: 
Invalid version found: 
Invalid version found: 
Invalid version found: 
Invalid version found: 

```

elektor-labs, 2016-12-28:
Upvote. I wasted a lot of time solving an "Invalid version found:" problem due to a library that had a space in the version string in its library.properties file. The offending string was displayed in full but without a clue as to where this string resided. Using aduino_debug I got off in the wrong direction because the error showed up right after "Preparing boards...". If only it displayed something like "Loading libraries...". Just adding "library" to the message "Invalid version found:" would already help a lot.

cmaglie, 2018-08-28:
Fixed by #7920

closed 2018-08-28 by cmaglie