arduino/Arduino #3220: ARM GCC not fully uninstalled (or re-installed) using boards manager.

damellis, 2015-05-25:
It looks like this symbolic link: ~/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/lib/gcc/arm-none-eabi/4.8.3/liblto_plugin.so (which points to itself?) doesn't get deleted when you uninstall the Arduino SAM boards in the boards manager. Then, if you re-install the SAM boards, ARM GCC doesn't get reinstalled, resulting in the following error when you try to compile a sketch:

Cannot run program "/Users/mellis/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" (in directory "."): error=2, No such file or directory

This is on Mac OS X (10.10.3) w/ Arduino 1.6.4 and a fresh ~/Library/Arduino15 directory. This may be the same as issue #3195.

cmaglie, 2015-05-25:
This may be already fixed with: #3141 
may you try with a nightly build?

ffissore, 2015-05-25:
Good catch! Indeed it should fix it

damellis, 2015-05-25:
Look like #3141 fixes it. Thanks!

closed 2015-05-25 by damellis