arduino/Arduino #3223: Related with the caret position during composing Korean characters - IDE 1.6.5

Myriads, 2015-05-25:
Normally(traditionally) the caret is always located at the right side of composing Korean character as like as the attached below picture of IDE 1.6.4.

[image]

But in IDE 1.6.5 the caret is located at the left side of composing Korean character:

[image]

ffissore, 2015-05-25:
When you type more than one char, do they get printed in the right order?

Myriads, 2015-05-25:
Yes, characters are inputted in the right order. Just caret is displayed at the left side of composing character.

ffissore, 2015-08-05:
@Myriads current hourly contains an updated version of the editor component. Can you give it a try and report if the problem persists? http://www.arduino.cc/en/Main/Software#hourly

Myriads, 2015-08-05:
Caret still has been displayed at the left side of composing Korean character.