arduino/Arduino #3228: Assigning a board to a sketch

Avamander, 2015-05-25:
Could someone please make it so that Arduino IDE automatically changes the board (and board model) based on the sketch that's opened. 

Having three-four different boards + board models, it is kinda tedious to select the board every time I open the sketch so something like "//BOARD=mega" that is added to the sketch and IDE automatically changes the board type would be really nice.

dmjlambert, 2015-06-12:
I love this idea.  During saving of the sketch, if the IDE remembered the board, processor, port, and any other boards.txt menu selections the user made on the Tools menu when working on the sketch, and then re-loaded those selections when opening the sketch, that would be fantastic.  I imagine this may be a bit difficult to code since you can have multiple sketches open at one time.  The code would have to take into consideration which window is in the foreground and active.  Your idea of line in the sketch to indicate the board is kind of an IDE directive and interesting.

Chris--A, 2015-06-13:
This is a good idea, but shouldn't be limited to just the board/port/etc

I personally would like a platform.txt style approach for each sketch (optional file). Comments in the code can be removed by unknowing users.

Sometimes I want to test with C++11 instead of C++98 or use some of the many other compiler/linker options, which cannot be changed without hard restarting the IDE.

ffissore, 2015-09-17:
Duplicate of #1481. Wontfix. Arduino Create will provide a similar feature

closed 2015-09-17 by ffissore

Avamander, 2016-03-13:
Would it be possible that this would be reconsidered at least partially?

romandubrov, 2017-04-12:
Bump!
I'd also like to have per-project settings in Arduino IDE because I'm completely tired with switching between my 4 boards. I like the online idea of Create but I need this feature locally because I store my projects and libs on Bitbucket. 
If you guys feel uncomfortable with storing settings in project.ino file - may be you'll like an idea of having separate optional file like project.settings?

JesseBusman, 2019-10-22:
To help anyone with this problem: If you start two arduino processes (by double-clicking your arduino icon twice) instead of two windows (by opening another file from an existing window) you can select different boards in the two different windows. (only if they run under separate procesess!)

It would still be awesome to not need this workaround and have the board selection saved in the .ino :)

dsyleixa, 2020-05-09:
IMO the most simple way would be to hardcode-config the targeted board type by configuration in Sketch (e.g. #pragma config...) such as
#pragma config board="Nano"
#pragma config board="NODE_MCU 1.0"
#pragma config board="Adafruit Feather ESP32"