arduino/Arduino #3231: Little question about (A>B)?A:B;

pierrotm777, 2015-05-25:
I try to play with an lcd I2C and try the example that follow!
      switch(StringToPrint)
      {
        case 0:lcd.print(F("Aux.:"));readAuxTypeLCD();break;
        case 1:lcd.print(F("Sync.:"));lcd.print((synchroIsActive == true)?"On  ":"Off ");break;
        case 2:lcd.print(F("T:"));lcd.print(GetTemp());lcd.print("\x06");lcd.print("C");break;
#ifdef EXTERNALVBATT
//        case 3:lcd.print(F("V:"));lcd.print((externalVBatt == true)?GetExternalVoltage():0.0);lcd.print("V  ");break;
        case 3:lcd.print(F("V:"));lcd.print(GetExternalVoltage());lcd.print("V  ");break;
#endif
      }

The case 3 commented don't run but the last run:
The code that read the external voltage is:
float GetExternalVoltage()
{
  externalVBatt = true;
  /*
  http://skyduino.wordpress.com/2012/08/09/arduino-mesurer-la-tension-dun-batterie/
  Celui ci a ete calcule pour permettre la mesure d'une plage de tension variant entre 0v et 20v.
  La formule classique d'un pont diviseur de tension est la suivante (je passe la partie theorie, loi des mailles, etc) :
  Vs = Vin * (R2 / (R1 + R2))
  Ici R1 = 3300 ohms (3k3) et R2 = 1100 ohms (1k1), on obtient donc pour Vin = 20v (maximum admissible) :
  Vs = 20 * (1100 / (3300 + 1100)) = 20 * (1100 / 4400) = 20 * 0.25 = 5
  Vs = 5v pour Vin = 20v, soit le maximum que peut mesurer le convertisseur analogique -> numerique de l'arduino (alimentation en 5v bien sur).

        +20V maxi
       -+-
        |
----.   # R1=3k3 = Orange Orange Rouge Or
    |   |
  A3|---+
    |   |        
NaNo|   # R2=1k1 = Marron Marron Rouge Or 
    |   |        
    |   |        
    |  -+-       
    |  GND      
----'        
  */  
  static float coeff_division = 4.0;
  /* Mesure de la tension brute */
  uint16_t raw_bat = analogRead(BROCHE_BATTEXT); //attention a I2C qui utilise les pins A4 et A5  
  /* Calcul de la tension reel */
  float real_bat = ((raw_bat * (5.0 / 1024)) * coeff_division);
//  /* Affichage */
//  Serial << real_bat << endl;
  return (real_bat);
}

Thanks for your explain !!!

ffissore, 2015-05-25:
I've edited your message so that code is properly rendered

pierrotm777, 2015-05-25:
Cool , thanks

cmaglie, 2015-05-25:
It seems that part of your message has been lost somewhere, so I don't understand exactly where's your issue.

BTW looks like an error in your sketch not in the LCD library, so you should probably open a discussion in the Arduino Forum http://forum.arduino.cc/ because this site is for tracking and fixing problems in the Arduino software.

agdl, 2015-06-05:
@pierrotm777 Waiting for feedback for too long. Reopen it if needed

closed 2015-06-05 by agdl