arduino/Arduino #3233: Wire.write() followed by Wire.requestFrom() results in inconsistent behavior when multiple bytes are transmitted

mjkrpo, 2015-05-25:
### Brief Description

I have been working with I2C communications lately, and have stumbled upon a situation where slave behavior becomes reproducibly erratic.

Steps:
 1. Master sends multiple bytes to one or more slaves (this problem ONLY occurs when 2+ bytes are sent).
 2. Master requests at least one byte from a slave (this problem ONLY occurs if data is requested).

### Environment

IDE: v1.6.4, though I've seen this in v1.0.6

Hardware: I used 1 Freeduino and 2 Ardweenys from Solarbotics.  All are Arduino clones (Uno) using the ATMega328.

Setup: Freeduino (master) connected to both Ardweeny slaves via I2C.  Both the SDA and SCL lines have 4.7k pullup resistors to 5 V

### Detailed Description

I had a very large project where I first ran into this problem.  In an effort to reduce the complexity, I have created a simpler setup for demonstration purposes.

The situation: I have a PC that is communicating via the Serial port to the Master Arduino.  I have an interface, written using Processing, that allows me to set the state of LED-13 (on/off) on each of the Arduino slaves.  The Arduino master receives this information over the Serial port, and relays this information to the slaves via I2C.

  As expected, this code works beautifully:
  
    // Given the following:
    byte SLAVE_A_ADDRESS = 0x3D;
    byte SLAVE_A_ADDRESS = 0x3E;
    byte slave_A_data = ...
    byte slave_B_data = ...

    // Send LED on/off command to each slave
    Wire.beginTransmission( SLAVE_A_ADDRESS );
    Wire.write( slave_A_data );
    Wire.endTransmission(); delay( 50 );
  
    Wire.beginTransmission( SLAVE_B_ADDRESS );
    Wire.write( slave_B_data );
    Wire.endTransmission(); delay( 50 );
  
    // Retrieve LED status from slaves
    // Here I am just asking whether or not LED-13
    //  is indeed on.  This will be important later.
    Wire.requestFrom( SLAVE_A_ADDRESS, 1, true );
    byte slave_A_status = Wire.read();
    Wire.endTransmission(); delay( 50 );
  
    Wire.requestFrom( SLAVE_B_ADDRESS, 1, true );
    byte slave_B_status = Wire.read();
    Wire.endTransmission(); delay( 50 );

  Things go quite differently when I try to send multiple bytes, like this:

    // Given the following:
    byte SLAVE_A_ADDRESS = 0x3D;
    byte SLAVE_A_ADDRESS = 0x3E;
    byte slave_A_data[2] = ...
    byte slave_B_data[2] = ...
    
    // Send LED on/off command to each slave
    Wire.beginTransmission( SLAVE_A_ADDRESS );
    Wire.write( slave_A_data, 2 );
    Wire.endTransmission(); delay( 50 );
  
    Wire.beginTransmission( SLAVE_B_ADDRESS );
    Wire.write( slave_B_data, 2 );
    Wire.endTransmission(); delay( 50 );
    
    // Retrieve status from slaves
    // Here I am just asking whether or not LED-13
    //  is indeed on.  This will be important later.
    Wire.requestFrom( SLAVE_A_ADDRESS, 1, true );
    byte slave_A_status = Wire.read();
    Wire.endTransmission(); delay( 50 );
  
    Wire.requestFrom( SLAVE_B_ADDRESS, 1, true );
    byte slave_B_status = Wire.read();
    Wire.endTransmission(); delay( 50 );

  Using the previous code chunk, Slave B will ALWAYS turn its LED on,
  regardless of what command has been sent; and yet, if it was the
  'off' command that had been sent, the subsequent status request
  will report 'off'.  This is completely non-sensical considering that
  the relevant slave code looks like this:
  
    byte data_in = Wire.read();
    byte unused_byte = Wire.read();  // If two bytes are sent
  
    // Process command
    if ( ( data_in & 0x01 ) == 0x01 ) {
        digitalWrite( 13, HIGH );
    }
    else {
        digitalWrite( ONBOARD_LED_PIN, LOW );
    }

  Here's where it gets interesting.  If I flip the order in which I send data to the slaves, from:
   1. Send 2 bytes to slave A
   2. Send 2 bytes to slave B
   3. Request 1 byte from slave A
   4. Request 1 byte from slave B

   To,
   1. Send 2 bytes to slave B
   2. Send 2 bytes to slave A
   3. Request 1 byte from slave A
   4. Request 1 byte from slave B

   Then slave B responds normally, but slave A behaves erratically.

### Workaround

   I have been able to get around this issue by inserting communication to a dummy slave device:

   1. Send 2 bytes to slave A
   2. Send 2 bytes to slave B
   3. Send 2 bytes to a slave that doesn't exist
   4. Request 1 byte from slave A
   5. Request 1 byte from slave B
   
   Both slaves now behave normally.

stickbreaker, 2015-06-15:
Why are you using endTransmission() after a requestFrom()?
endTransmission() is only used in the sending of data [beginTransmission(),write(),endTransmission()]

When you call endTransmission() the Wiring library initiates a transmit to the address of the last Wire.beginTransmission(I2Caddr).  The data it transfers is what ever is in the buffer from the last activity.

I don't think that is what you wanted?

Chuck.

mjkrpo, 2015-06-15:
You are absolutely right Chuck; that was an oversight on my part.  I removed the extranneous endTransmission() calls and that fixed all of my problems.

Thanks, and I'm glad to see that it wasn't anything major.

sandeepmistry, 2015-10-06:
@stickbreaker thanks for helping out!

Closing as issue was caused by ```Wire.endTransmission()``` being used with ```Wire.requestFrom``` when it wasn't needed.

closed 2015-10-06 by sandeepmistry

tonton81, 2017-09-18:
sorry to fill the blind eyes but...

byte SLAVE_A_ADDRESS = 0x3D;
byte SLAVE_A_ADDRESS = 0x3E;
byte slave_A_data = ...
byte slave_B_data = ...

this doesnt seem right lol