arduino/Arduino #3234: Invalid signature file digest for Manifest main attributes

ShorTie8, 2015-05-26:
With the "Introducing GPG signature verification when loading main package_index" in 1.6.4

https://github.com/arduino/Arduino/commit/cf1879f948e0c1fa9ddd91f7348fbdd07fd870cb

I have not been able to get the IDE to run using the system files.
I need to use the system files because I'm trying to run the IDE on a Raspberrypi which is ARM and you offer no pre-compiled packages for the ARM.

Is there a easy way to disable the GPG checking ??

Thankz
Have A Great Day

ffissore, 2015-05-26:
If for "easy" you mean a switch to turn on and off, then no, no easy way

ShorTie8, 2015-05-26:
ya, 'a switch to turn on and off' would have been nice.
Would you be so kind to give some insight on how to get around it then ??

ffissore, 2015-05-26:
You already located the commit. You need to dig into the code and find your way out: `isSigned` function is what you're looking for

closed 2015-05-26 by ffissore

ShorTie8, 2015-05-26:
Thank you Sir
The insight basically did provide me with a simple button to turn it off.

Have A Great Day