arduino/Arduino #3238: CMD+SHIFT+ARROW should select text on Mac OS X

Lauszus, 2015-05-26:
I believe that the keyboard shortcut for selecting text is wrong on Mac OS X after the introduction of RSyntaxTextArea.

Currently the shortcut is FN+SHIFT+ARROW, but in Arduino 1.6.4 and below it was CMD+SHIFT+ARROW.

ffissore, 2015-05-27:
You mean current shortcut is ALT+SHIFT+ARROW right?

Lauszus, 2015-05-27:
No. ALT+SHIFT+ARROW should only select the next word and works as expected. CMD+SHIFT+ARROW should select the entire line.

Lauszus, 2015-05-27:
I just tried FN+SHIFT+ARROW and it seems like the standard is that it selects all text on the line and any text that follows. Let say I press FN+SHIFT+LEFT, then it selects the text to the left and all text above it.

Lauszus, 2015-05-27:
I just found this Wiki article which backs up my claim: http://en.wikipedia.org/wiki/Table_of_keyboard_shortcuts#Text_editing.

ffissore, 2015-05-27:
Please give a spin to #3239 as soon as @ArduinoBot has published a build

commit 0956f5da (2015-06-01): Merge pull request #3239 from ffissore/macosx-select-line

MacOSX: CMD+SHIFT-LEFT/RIGHT selects the entire line. Fixes #3238
commit dcbc8e21 (2015-06-01): MacOSX: CMD+SHIFT-LEFT/RIGHT selects the entire line. Fixes #3238
closed 2015-06-01 by ffissore