arduino/Arduino #3241: Latest release 1.6.5 - cannot run -> java.lang.UnsatisfiedLinkError

robertgregor, 2015-05-27:
Hello, 
I have used the latest build of Arduino with Windows 7 64 bits. When I tried to start, I have the following error:

c:\RH\arduino-1.6.5>arduino_debug
Loading configuration...
java.lang.UnsatisfiedLinkError: jnidispatch (/com/sun/jna/win32-amd64/jnidispatch.dll) not found in resource path
        at com.sun.jna.Native.loadNativeLibrary(Native.java:471)
        at com.sun.jna.Native.<clinit>(Native.java:91)
        at processing.app.windows.Advapi32.<clinit>(Advapi32.java:21)
        at processing.app.windows.Registry.getStringValue(Registry.java:115)
        at processing.app.windows.Platform.getSettingsFolder(Platform.java:194)
        at processing.app.BaseNoGui.getSettingsFolder(BaseNoGui.java:287)
        at processing.app.BaseNoGui.getSettingsFile(BaseNoGui.java:272)
        at processing.app.PreferencesData.init(PreferencesData.java:38)
        at processing.app.BaseNoGui.initParameters(BaseNoGui.java:946)
        at processing.app.Base.guardedMain(Base.java:153)
        at processing.app.Base.main(Base.java:133)

ffissore, 2015-05-27:
`processing.app.windows.Advapi32` has been deleted with #3194 Please download the hourly build again

robertgregor, 2015-05-28:
Hello, seems like the issue is still there. What was exactly the change?

Robert

ffissore, 2015-05-29:
You're using a 64 bit jvm. Arduino for windows works only with 32bit jvm. And indeed we bundle a 32 jvm

closed 2015-05-29 by ffissore