arduino/Arduino #3242: Document the WiFi status() function.

damellis, 2015-05-27:
It's used in the examples but I don't see it in the reference. We should document its possible return values too.

agdl, 2015-05-28:
@damellis here the web page i made
http://arduino.cc/en/Reference/WiFiStatus
Let me know if it is good for you

damellis, 2015-05-28:
Thanks! This looks like a good starting point. It would be helpful to provide more documentation about when the various statuses occur. (This isn't really about related to the function itself, but I'm not sure where else you'd document it.) For example, WL_CONNECT_FAILED, WL_CONNECTION_LOST, WL_DISCONNECTED seem to refer to a result of an action, rather than an ongoing status. Does the status revert back from those three to WL_IDLE_STATUS? If so, when? Or does WL_IDLE_STATUS only exist at the very beginning, when the shield has been connected but nothing else has happened yet? Etc.

agdl, 2015-05-29:
@damellis Take a look now

damellis, 2015-05-29:
Looks good. Clarifying that the IDLE status is a temporary one makes
everything else much clearer. You could maybe make the "CONNECTED"
description more like the old one (e.g. "when connected to a wifi network")
but that's a minor point.

agdl, 2015-06-01:
Done thank you

closed 2015-06-01 by agdl