arduino/Arduino #3248: Arduino 1.6.4 crashes/closes before it opens

questuk, 2015-05-28:
Hi,

A few of us are having problems with opening Arduino IDE 1.6.4.on different operating systems.

Here's the link to the forum responses, http://forum.arduino.cc/index.php?topic=325527.0

I see the loading screen and when it states "initializing packages" it just closes. No crash report, no indication of something going wrong, just closes. I've tried uninstalling the IDE and reinstalling, but nothing seems to make a difference. 

I tried running arduino_debug.exe.This is the output.

C:\Program Files\Arduino>arduino_debug.exe
Loading configuration...
Initializing packages...
java.lang.NullPointerException
       at cc.arduino.contributions.packages.ContributedToolReference.resolve(ContributedToolReference.java:43)
       at cc.arduino.contributions.packages.ContributedPlatform.resolveToolsDependencies(ContributedPlatform.java:76)
       at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:92)
       at processing.app.BaseNoGui.initPackages(BaseNoGui.java:600)
       at processing.app.Base.<init>(Base.java:271)
       at processing.app.Base.guardedMain(Base.java:215)
       at processing.app.Base.main(Base.java:128)

C:\Program Files\Arduino>

Regards

Gary

ffissore, 2015-05-29:
Do you still have your arduino15 folder? It looks like one of the downloaded `.json` file is empty or invalid. Do you have a chance to upload the entire arduino15 folder somewhere I can download it from?

per1234, 2015-05-29:
I'm not sure if it is the same issue but you can get a similar result by adding:
http://arduino.esp8266.com/package_esp8266com_index.json,https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
to Tools > Preferences > Additional Boards Manager URLs as reported on the forum: http://forum.arduino.cc/index.php?topic=324095.0 and https://github.com/damellis/attiny/issues/66
Either json by itself causes no problems and I can use:
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json,https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
with no problems. I looked at both files and haven't found anything obvious.

ffissore, 2015-05-29:
Maybe I guessed the source of the problem: esp8266 latest package is 1.6.4-673-g8cd3697. You've probably installed an older version and they removed that old version from the file. The IDE finds an installed esp8266 core but can match it with any of the listed ones.
I'll code a workaround that will ignore unrecognized cores. Please tell esp guys to refrain from deleting old cores

per1234, 2015-05-29:
It happens even with no esp8266 versions installed. I add the URLs to preferences, then open Boards manager, it downloads the json files, then shows java.lang.NullPointerException. Neither of the packages show in Boards Manager, they are not installed. The console shows:
```
java.lang.NullPointerException
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$3.run(ContributionManagerUI.java:158)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributedToolReference.resolve(ContributedToolReference.java:43)
	at cc.arduino.contributions.packages.ContributedPlatform.resolveToolsDependencies(ContributedPlatform.java:78)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:100)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:612)
	at processing.app.Base$12.onIndexesUpdated(Base.java:1214)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$3.run(ContributionManagerUI.java:156)
	... 1 more
java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributedToolReference.resolve(ContributedToolReference.java:43)
	at cc.arduino.contributions.packages.ContributedPlatform.resolveToolsDependencies(ContributedPlatform.java:78)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:100)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:612)
	at processing.app.Base.openInstallBoardDialog(Base.java:1229)
	at processing.app.Base.access$100(Base.java:80)
	at processing.app.Base$13.actionPerformed(Base.java:1249)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```
When I close the IDE and try to restart I get the splash screen but the IDE won't start until I delete either of the json files.

ffissore, 2015-05-29:
Then maybe it's not the esp index the source of the problem, maybe it's the attiny one

per1234, 2015-05-29:
attiny is not installed either and https://adafruit.github.io/arduino-board-index/package_adafruit_index.json,https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json works fine. I'm not saying the json files don't have a problem, but the problem is not that I have an old version installed that has been removed from the json.

ffissore, 2015-05-29:
I see. Indeed further tests (with latest hourly build) show no problem. Can you give a try to hourly as well? http://www.arduino.cc/en/Main/Software#hourly I can't reproduce the bug :(

per1234, 2015-05-29:
Same result, I'm using Windows 7 64 bit. Here's my process:
- Install latest hourly
- Delete all files related to the two packages from Arduino15 folder and sketchbook/hardware folder
- open IDE
- Add http://arduino.esp8266.com/package_esp8266com_index.json,https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
to Tools > Preferences > Additional Boards Manager URLs
- Tools > Board > Boards Manager -  it downloads the json files, then shows java.lang.NullPointerException
- Console shows error messages shown  in my previous comment.
- Close IDE
- start IDE - splash screen shows but IDE doesn't open
- arduino_debug.exe shows the same error messages shown in my previous comment
- delete either json file from Arduino15 folder
- start IDE - IDE opens fine but running Boards Manager has the same result as before.

questuk, 2015-05-29:
On 29/05/2015 10:15, Federico Fissore wrote:
>
> Do you still have your arduino15 folder? It looks like one of the downloaded 
> |.json| file is empty or invalid. Do you have a chance to upload the entire 
> arduino15 folder somewhere I can download it from?
>
> —
> Repy to this email directly or view it on GitHub 
> <https://github.com/arduino/Arduino/issues/3248#issuecomment-106753030>.
>

Hi Federico,

Here is the attached folder you asked for, this is at the time or just  after 
the crash.

I was on this page https://github.com/esp8266/arduino

and followed the section *'Installing with Boards Manager**' *to the letter

I I had 2 or 3 other items I had successfully added in the board manager URL and 
then finally this last one, all the items added, were separated with a comma and 
a space.

Regards

Gary

ffissore, 2015-05-29:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly
Problem was that indexes like that of codebender missed tools: it doesn't need tools, but we gave for granted that any index would supply its tools, and we were wrong

closed 2015-05-29 by ffissore

per1234, 2015-05-29:
ffissore is there a workaround for making package indexes without tools compatible with 1.6.4? I am the author of the attiny json and the ariadne json also(I think that is what you mean by codebender) and a couple of others and I'd like to fix the files if possible. I tried adding `"tools": [{}]` but that didn't work.

Thanks so much for your work on this fix and Boards Manager in general!

per1234, 2015-05-29:
Disregard that last question I figured it out. I just needed to add "tools": [] and it works.

ffissore, 2015-05-29:
Yes indeed

supuflounder, 2017-09-16:
I am having the same problem with 1.8.4, over two years later.  It gets to where it says "Starting" then softly and silently vanishes away..  The above solutions do not work.

supuflounder, 2017-09-16:
I figured out how to run it in a console, and got the following report:

Josephs-MacBook-Air:~ flounder$ /Applications/Arduino-3.app/Contents/MacOS/Arduino ; exit;
java.lang.IllegalArgumentException: Font size must be > 0
	at org.fife.ui.rtextarea.RTextAreaBase.setFont(RTextAreaBase.java:986)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setFont(RSyntaxTextArea.java:2510)
	at processing.app.EditorTab.applyPreferences(EditorTab.java:322)
	at processing.app.EditorTab.<init>(EditorTab.java:110)
	at processing.app.Editor.addTab(Editor.java:1672)
	at processing.app.Editor.createTabs(Editor.java:1645)
	at processing.app.Editor.handleOpenInternal(Editor.java:1944)
	at processing.app.Editor.<init>(Editor.java:340)
	at processing.app.Base.handleOpen(Base.java:833)
	at processing.app.Base.restoreSketches(Base.java:518)
	at processing.app.Base.<init>(Base.java:456)
	at processing.app.Base.main(Base.java:140)

per1234, 2017-09-16:
@supuflounder please see https://github.com/arduino/Arduino/issues/6716 for an explanation of the problem and solutions. Although it has similar symptoms to this issue, your problem has a different cause.

commit 1b7574a8 (2015-05-29): Boards Manager: contributions MAY miss tools, but we missed to consider this case. Fixes #3248