arduino/Arduino #3251: [DOC] It isn't clear which image formats are supported by official TFT module

q2dg, 2015-05-29:
Maybe a simple sentence should be appened in http://www.arduino.cc/en/Reference/TFTLoadImage indicating the correct format (o formats?). 
*Are only BMP 24 bits supported? Or BMP with other color-depth are too? 
*Are JPG or PNG photos impossible to load? 
*In any case, must photos be located at root of SD?
They are pretty simple question, but I think are fundamental to any novice.

agdl, 2015-05-29:
@q2dg I added the needed informations. Let me know if it is good and close the issue if so. Thank you!

q2dg, 2015-05-29:
Yeah, for me it's fine!! Thanks a lot!!!

PD: Well, maybe "The TFT library has the ability to read .bmp files off **the root of** a SD card and display them on the screen" would be even better, don't you think?

closed 2015-05-29 by q2dg

agdl, 2015-05-29:
done thank you!