arduino/Arduino #3256: Green title bar button should enter full-screen mode on OS X (10.10 and later).

damellis, 2015-05-29:
This is the new standard behavior for most OS X apps and I think it would be nice to follow the convention.

damellis, 2015-06-01:
Thanks for working on this!

It doesn't seem to be working properly, though. Clicking the green button now slides the window over onto its own "desktop" as you'd expect, but it doesn't actually expand to fill the whole screen. 

[image]

It then seems to stop responding to user input. If you Command-Tab to a different application and then back to Arduino, there are two copies of the window, one full-screen and one the original size, but neither seems responsive to any input. 

[image]

Again, this probably isn't a priority, even for a OS X fan like me (but you probably don't want to leave it as it is now).

ffissore, 2015-06-01:
Each window goes full screen on its own. I've tested this on a Yisemite and
it works as expected

ffissore, 2015-06-01:
Commit reverted. Waiting for mac user contribution

damellis, 2015-06-01:
Weird, not sure why it didn't work for me if it did for you.

damellis, 2015-06-04:
Looks like the problem may have been that I was building Arduino with an older version of Java (1.7.0_60). When I build with the latest version (1.8.0_45) this works as expected (and the fonts look much better too!). Any other tests I should do?

commit e3755719 (2015-06-01): MacOSX: fullscreen support. Fixes #3256
commit e0821130 (2015-06-01): Revert "MacOSX: fullscreen support. Fixes #3256"

This reverts commit e375571945900c9fa629fe72d0bba46b9b08c176.