arduino/Arduino #3259: Arduino should ask before silently creating the folder "sketchbook" in the home directory

mghansen256, 2015-05-30:
When starting the Arduino UI (version 2:1.0.5+dfsg2-4) for the first time on Linux, a folder "sketchbook" is created in the home directory. IMHO the program should first ask the user whether it may create the folder there or elsewhere, in order not to clutter a user's home directory with unknown folders.

tytower, 2015-05-30:
One thing is for sure . If its unknown to you now it won't be for long if you use Arduino

sspence65, 2015-05-30:
Where else would you have it create sketchbook? It's not an unknown folder, it's where all your sketches and 3rd party libraries are held.

ffissore, 2015-06-01:
Wontfix. All available documentation uses the *default sketchbook location*

closed 2015-06-01 by ffissore