arduino/Arduino #3260: V1.64 IDE Run-Time Problem under WINXP-32-BIT

emrtech, 2015-05-31:
I sometimes use an older WINXP32 PC in my lab and discovered that the V164 version IDE suddenly disappears after start up after a few+ seconds for no apparent reason. Sometimes it works for a minute or so and then dissapears. I was able to compile a project once. V161 works properly on the XP machine. I have used V164 on WIN7X64 and WIN8.1X64 machines without any problems.

I just want to report this problem and I am not expecting any resolution to this problem as WINXP is no longer an active OS.

Best Regards,
Gerhard

ffissore, 2015-06-01:
@emrtech next time please start the IDE by opening a command prompt, `cd` into IDE installation folder (like `cd "c:\program files\arduino"`) and start the IDE with `arduino_debug.exe`. When it will die, it will print an error message: paste it here

ffissore, 2015-06-05:
Made some tests with a Windows XP SP3. Found and fixed a bug (see e9d66015a4490de03292e2a6872c4bec69d6e151). Closing for lack of feedback. Reopen if needed

closed 2015-06-05 by ffissore

emrtech, 2015-06-05:
Dear Federico,

Thank you for finding and fixing the problem. Your speed on this left  
me in the dust. I didn' t have time yet to get you the information due  
to my involvement with my sick mother. I will test your changes  
hopefully this weekend.

Best Regard,
Gerhard