arduino/Arduino #3261: CLI sketchbook.path pref 1.6.4

jacobrosenthal, 2015-05-31:
At least for me, Arduino 1.6.4 under OSX 10.10.3, Im unable to utilize the sketchbook.path pref. 

```bash
/Applications/Arduino.app/Contents/MacOS/Arduino --pref sketchbook.path=/Users/jacobrosenthal/Documents/firmware-pinoccio --board pinoccio:avr:atmega256rfr2 --upload /Users/jacobrosenthal/Documents/firmware-pinoccio/libraries/pinoccio/examples/Bootstrap/Bootstrap.ino --port /dev/tty.usbmodem1411
```

Fails with "pinoccio: Unknown Package"

However if I save the new directory in sketchbook location of the Arduino GUI then run the arduino cli without overloading it works fine.

FWIW I also tested with non system libraries (my Goertzel library) but system hardware (arduino:avr:uno) and I get something like:
detect.ino:32:22: fatal error: Goertzel.h: No such file or directory
So its not board architectures, its the entire sketchbook path.

ffissore, 2015-06-01:
Works fine with latest hourly. Could it be the order of the params? Please try with: 
`/Applications/Arduino.app/Contents/MacOS/Arduino --pref sketchbook.path=/Users/jacobrosenthal/Documents/firmware-pinoccio --board pinoccio:avr:atmega256rfr2 --upload --port /dev/tty.usbmodem1411 /Users/jacobrosenthal/Documents/firmware-pinoccio/libraries/pinoccio/examples/Bootstrap/Bootstrap.ino`

jacobrosenthal, 2015-06-01:
Thanks for the triage! No difference with that command. Ill try to grab head and maybe a version back and see if its there as well.

jacobrosenthal, 2015-06-02:
Didnt work for me with the hourly. You're on osx 10.10.3?
/Users/jacobrosenthal/Downloads/Arduino.app/Contents/MacOS/Arduino --pref sketchbook.path=/Users/jacobrosenthal/Documents/firmware-pinoccio --board pinoccio:avr:atmega256rfr2 --verify /Users/jacobrosenthal/Documents/firmware-pinoccio/libraries/pinoccio/examples/Bootstrap/Bootstrap.ino

matthijskooijman, 2015-06-03:
@jacobrosenthal, I presume that the pinoccio board files are stored inside the custom sketchbook path? If so, the problem is probably that the pinoccio board files aren't loaded by the time the --board option is processed?

Did this used to work? I have been using the same commandline, but I probably had the pinoccio board files *also* in my regular sketchbook path...

ffissore, 2015-06-03:
I confirm the bug. Packages (hardware folder content) are parsed **before** the `sketchbook.path` pref is applied, making it useless. I'll work on it

jacobrosenthal, 2015-06-03:
Yay! Thanks guys.

ffissore, 2015-06-05:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly
However, FYI, I'm now getting an error
```
federico@smilzo:~/materiale/works_Arduino/Arduino/build (master)$ linux/work/arduino --pref sketchbook.path=/home/federico/Arduino2 --board pinoccio:avr:pinoccio:cpu=atmega256rfr2 --verify /home/federico/Arduino2/libraries/Pinoccio/examples/Bootstrap/Bootstrap.ino --verbose
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Uso la libreria SPI nella cartella: /home/federico/materiale/works_Arduino/Arduino/build/linux/work/hardware/arduino/avr/libraries/SPI 
Uso la libreria Wire nella cartella: /home/federico/materiale/works_Arduino/Arduino/build/linux/work/hardware/arduino/avr/libraries/Wire 
Uso la libreria Pinoccio nella cartella: /home/federico/Arduino2/libraries/Pinoccio 

/home/federico/materiale/works_Arduino/Arduino/build/linux/work/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega256rfr2 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_PINOCCIO -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/home/federico/materiale/works_Arduino/Arduino/build/linux/work/hardware/arduino/avr/cores/arduino -I/home/federico/Arduino2/hardware/pinoccio/avr/variants/pinoccio -I/home/federico/materiale/works_Arduino/Arduino/build/linux/work/hardware/arduino/avr/libraries/SPI -I/home/federico/materiale/works_Arduino/Arduino/build/linux/work/hardware/arduino/avr/libraries/Wire -I/home/federico/Arduino2/libraries/Pinoccio/src /tmp/build2687018480008254479.tmp/Bootstrap.cpp -o /tmp/build2687018480008254479.tmp/Bootstrap.cpp.o 
In file included from /home/federico/Arduino2/libraries/Pinoccio/src/Scout.h:12:0,
                 from Bootstrap.ino:11:
/home/federico/Arduino2/libraries/Pinoccio/src/Pinoccio.h:21:31: fatal error: lwm/sys/sysConfig.h: No such file or directory
 #include "lwm/sys/sysConfig.h"
                               ^
compilation terminated.
Errore durante la compilazione
```

ffissore, 2015-06-05:
Also, take a look at the --board parameter
It's `--board pinoccio:avr:pinoccio:cpu=atmega256rfr2` for boards that use menu options like `cpu`

closed 2015-06-05 by ffissore

matthijskooijman, 2015-06-06:
@ffissore, that error is because you're missing some libraries. I think the bug reported here is fixed. @jacobrosenthal, can you confirm in your setup (with all libraries installed) and close this ticket?

jacobrosenthal, 2015-06-06:
Yep, looks good on my end.

commit 84d10a8e (2015-06-05): Some CLI args are parsed before initializing packages. --board argument is parsed after. Fixes #3261