arduino/Arduino #3263: Unofficial boards list issue

pierrotm777, 2015-05-31:
The ide 1.6.4 and 1.6.5 return me an java error with these two json links:
https://raw.githubusercontent.com/carlosefr/atmega/master/package_carlosefr_atmega_index.json
http://rfduino.com/package_rfduino_index.json

The problem is than after this issue , the IDE wan't run . I need to delete all json file !
I have found a solution:
Don't quit the IDE after the error , edit the preferences file (delete the bad json link) and launch again the board manager !

per1234, 2015-05-31:
This sounds like the same problem as #3248. https://raw.githubusercontent.com/carlosefr/atmega/master/package_carlosefr_atmega_index.json was missing the ***tools*** section. I have patched that file and tested Boards Manager install of those links under Arduino 1.6.4 and Hourly Build 2015/05/29 04:46 without any problems. Could you please try it again @pierrotm777 and see if you still have the problem.

ffissore, 2015-06-01:
This is fixed in 1b7574a8b9ac5cbbac4809bdd0a6a27014d7ea26. See #3248

closed 2015-06-01 by ffissore