arduino/Arduino #3264: Missing documentation for String() constructors

dwwolfe, 2015-05-31:
Issue #1297 added support for constructing a String object with a float/double.  The documentation for string constructors does not reflect the ability to do this:

http://www.arduino.cc/en/Reference/StringConstructor
http://www.arduino.cc/en/Tutorial/StringConstructors

I've wasted a lot of time trying to get around this issue using dtostrf() and writing custom functions so it would have been much nicer to immediately find it on the reference pages!  Thanks

agdl, 2015-06-01:
@dwwolfe it is online now. If it is good for you please close this issue

dwwolfe, 2015-06-01:
The reference page looks good, thanks.  I'm not sure whether it is worth updating the tutorial page as well?  I will let that up to you but a google search for arduino string constructors turns up both pages that I cited so a user could still land on a page without any mention of float/double support.  If those tutorial pages are not typically updated then I can just close the issue

agdl, 2015-06-01:
They will be updated automatically because they dump the code from github
so now that the PR is merged they will go online as well

dwwolfe, 2015-06-01:
Very good, thanks again

closed 2015-06-01 by dwwolfe