arduino/Arduino #3268: Serial monitor won't open after "Error opening serial port 'COM12'. (Port not found)" under latest hourly build

per1234, 2015-06-01:
Under Windows 7 64 bit using Arduino IDE hourly build 2015/06/01 09:34:
Once I get the `Error opening serial port 'COM12'. (Port not found)` error the serial monitor won't open any port until the IDE is restarted. This happens with any com port number. I don't have this problem under 1.6.4. It is a bit difficult to get the error message. I can get it sometimes by:
- Plug in the board to USB
- Select a different port from the one the board is on
- Select the port the board is on
- Unplug the board from USB
- Try to open the serial monitor.
- Plug the board back in to USB
- Verify the port shows in the ports menu
- Open serial monitor. Nothing happens. The same will happen for boards on a different port.

It may take a few tries to get the error message. I do not have this problem after the more common `Board at COM12 is not available` error.

ffissore, 2015-06-01:
Related to #3255. Fix under way

ffissore, 2015-06-01:
Should be fixed in latest hourly build http://www.arduino.cc/en/Main/Software#hourly Can you give it a try?

per1234, 2015-06-01:
Hourly Build 2015/06/01 12:19, still same problem.
This has happened for me with FTDI connected to Pro Mini, Mega2560, and Pro Micro so it may be different from #3255 which sounds like it is related to Due/Leonardo.

per1234, 2015-06-01:
I can still upload sketches after I get the error and serial monitor won't open so the serial port is still working, just not the serial monitor.

closed 2015-06-01 by ffissore

smeezekitty, 2017-07-31:
I'm still experiencing this bug in 1.8.2

commit b77f5f73 (2015-06-01): If opening serial monitor causes an error, set it closed. Fixes #3268