arduino/Arduino #3272: Unable to connect

Phantomarduino, 2015-06-01:
I have a Robo 3D R1 printer.

I went to the instructions of the printer to see how to update the firmware.
Downloaded "arduino-1.6.4-windows.exe"
Downloaded "autolevel-7-17-516inch.zip"
Created a directory "documents/arduino/sketchbook"
Unzipped the file to the above directory
Ran arduino
Chose the option upload and said it was ready.
Now, I can´t connect with the software that drives the printer to the printer

But now mattercontrol can´t connect to the printer.

Did I do something wrong?
Thanks

ffissore, 2015-06-01:
This repo issues are for bug related to the Arduino IDE. Your enquiry looks much more like a support request. Your should ask support to your Robo 3D R1 printer vendor/supplier. Should you find an actual bug, don't hesitate to open a new issue

closed 2015-06-01 by ffissore

Phantomarduino, 2015-06-01:
If the arduino program blocks the possibility to connect with the arduino board it seems to me that its an arduino issue