arduino/Arduino #3276: java.lang.UnsatisfiedLinkError: C:\Download\Arduino\build\windows\work\lib\AStylej.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform

robertgregor, 2015-06-02:
Hello, I have Windows 7 64 bit platform. The ant build is successfull, but when I try to start it, I have the following error:

c:\Download\Arduino\build\windows\work>arduino_debug
Loading configuration...
Initializing packages...
Preparing boards...
Starting...
java.lang.UnsatisfiedLinkError: C:\Download\Arduino\build\windows\work\lib\AStyl
ej.dll: Can't load IA 32-bit .dll on a AMD 64-bit platform
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1937)
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1822)
        at java.lang.Runtime.load0(Runtime.java:809)
        at java.lang.System.load(System.java:1086)
        at cc.arduino.packages.formatter.AStyleInterface.<clinit>(AStyleInterfac
e.java:42)
        at cc.arduino.packages.formatter.AStyle.<init>(AStyle.java:56)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)

        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstruct
orAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingC
onstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:422)
        at java.lang.Class.newInstance(Class.java:442)
        at processing.app.Editor.createToolMenuItem(Editor.java:1041)
        at processing.app.Editor.addInternalTools(Editor.java:1064)
        at processing.app.Editor.buildToolsMenu(Editor.java:776)
        at processing.app.Editor.buildMenuBar(Editor.java:549)
        at processing.app.Editor.<init>(Editor.java:240)
        at processing.app.Base.handleOpen(Base.java:879)
        at processing.app.Base.handleOpen(Base.java:864)
        at processing.app.Base.handleOpen(Base.java:860)
        at processing.app.Base.handleNew(Base.java:748)
        at processing.app.Base.<init>(Base.java:470)
        at processing.app.Base.guardedMain(Base.java:224)
        at processing.app.Base.main(Base.java:135)
C:\Download\Arduino\build\windows\work\lib\AStylej.dll: Can't load IA 32-bit .dl
l on a AMD 64-bit platform
Cannot load native library c:\Download\Arduino\build\windows\work\lib\astylej.dl
l
The program has terminated!

I have used latest git clone...

Chris--A, 2015-06-03:
Do you have the 64 bit JDK/JRE installed?

You need the 32-Bit versions. I got everything working with JDK 7 32-Bit

ffissore, 2015-06-03:
Indeed. Windows dev must use a 32bit jvm because that is the jvm we bundle with the IDE

closed 2015-06-03 by ffissore

bc-185, 2020-12-05:
> Indeed. Windows dev must use a 32bit jvm because that is the jvm we bundle with the IDE

It is more like the program is made for 32 bit Java, because I can run the program with later 32 bit Java versions. (like Java 11).