arduino/Arduino #3277: Input/Output Pin Functions

HankGG, 2015-06-03:
Arduino is fantastic and easy to use.  Congrats.  Since most of the controls to the outside world are thru Input/Output pins, maybe you could increase Arduino's command words to:
Din (pin)                             same as     int digitalRead (pin)
Dout (pin, value)                 same as     digitalWrite (pin, value)
Ain (pin)                              same as     int analogRead (pin)
Aout (pin, value)                 same as     int analogWrite (pin, value)

This is taken from process-control language and is easier to type in.  You would still leave the original commands for older sketches but also accept the newer commands going forward.

In future, Frequency functions ??
Fin (pin)
Fout (pin, value)

tamul, 2015-06-11:
I feel as though part of Arduino's ease of use is the more or less self explanatory functions. When function names start to get more cryptic, the codebase definitely gets harder to keep track of.

per1234, 2015-06-11:
Maybe using a library to add these functions would be a better option. @HankGG if that sounds like something you would find useful but don't know how to write one yourself let me know and I'll upload it to Github.

HankGG, 2015-06-11:
Thanks for the input guys.  Firstly, I am not a programmer but a hardware guy (that's the problem).  I just used the simplest software programs for debugging hardware.  Just thought that life could be easier because sometimes we want to put too much in a command.  For example 'cout' is used in C++  to print to the monitor.  Use it once and you know it.
It could have used "SerialOUTPUTMonitor" (x,y)  // just pulling your leg
Another example is 'DelayMicroseconds', I prefer 'DelayuSec' // not pulling your leg
If you are happy with the syntax, who am I to argue, I can live with it.
I'm a guy who programmed a Z8 with Hex Code (not Assembly Language) without using labels for jumps, all written and calculated by hand.  Didn't have a development system.  (Don't change your program too often).  Have a US patent for the project design.
Also when PALs came out, programmed the chip by working out 3 pages of Boolean algebra by hand.  Programmer too expensive at the time.  PAL worked 100% first time.
So I really enjoying playing around with Arduino. what a pleasure, just need to learn coding.  // by your rules.   :-) Take care.

VogonJeltz, 2015-07-06:
I'm not quite sure if I understood you right, but about
Din (pin) same as int digitalRead (pin)
Dout (pin, value) same as digitalWrite (pin, value)
Ain (pin) same as int analogRead (pin)
Aout (pin, value) same as int analogWrite (pin, value)

what about:

#define Din(pin)  digitalRead(pin)
#define Dout(pin, value)  digitalWrite(pin, value)
#define Ain(pin)  analogRead(pin)
#define Aout(pin, value)  analogWrite (pin, value)

??
is this what you meant?

matthijskooijman, 2015-07-06:
As for including things like these in the official Arduino core, that won't happen. Because:
 - The shorter names are ambiguous and confusing, Arduino strives to be clear and readable.
 - Having multiple names for the same functions is confusing as well
 - Short names like proposed have a big chance to conflict with the code in people's sketches (*especially* if defines are used).
 - Every function name added to Arduino must be supported by third-party cores as well. With no real gain, this is a lot of effort.

Hence I'll close this issue and invite you to implement your things in a custom header file or library (or find someone else willing to do so, some suggestions have been given already).

closed 2015-07-06 by matthijskooijman