arduino/Arduino #3281: 1.6.4 installed June 03, Tools doesn't list any Ports while Edison plugged into Windows 7

gblaker, 2015-06-04:
On June 03, 2015, downloaded and used the Arduino site's windows installer for Arduino IDE 1.6.4 (from the previous night's build), to replace a previous 1.6.4 installation (hoping to cure the problem)
Used Tools-> Board Manager to remove and reinstall the associated Intel i686 Boards package (hoping to cure the problem).
Windows is 64-bit Windows 7

With an Edison on an Arduino Carrier board, connected to my PC by only the two microUSB ports (the selector switch is set to choose using them), when I start the Arduino IDE, the Tools->Port entry is always greyed out.  This is the symptom.

I can still transfer code into the Edison-Arduino, but I can't use the IDE's Serial Monitor.  This is the effect/problem.

In this situation, Device manager shows these ports exist:
 - Communications Port (COM1)
 - Intel Edison USB Composite Device (COM11)
 - Intel Edison USB Virtual Comm Port (COM10)
 - USB Serial Port (COM8)

Here is a workaround (but I'm getting really tired of using it...).

If I disconnect the Edison, Windows' Device Manager shows these ports:
 - Communications Port (COM1)

If I start the Arduino IDE while the Edison is disconnected, the Tools_>Port entry is usable, and shows the COM1 port.

If I then reconnect my PC to the Edison, Windows' Device manager shows these ports again:
 - Communications Port (COM1)
 - Intel Edison USB Composite Device (COM11)
 - Intel Edison USB Virtual Comm Port (COM10)
 - USB Serial Port (COM8)

Checking the IDE's Tools->Ports entry now, shows me COM1, COM8 and COM10 choices.

I can select COM10 and resume using the IDE's serial monitor with the Edison-Arduino.

ffissore, 2015-06-04:
I'm not sure I can solve the problem. Can you try using latest hourly build and see if that too is shows your problem? http://www.arduino.cc/en/Main/Software#hourly

gblaker, 2015-06-05:
Thank you ffissore - I'm chasing a different problem today - When I am able to test using the latest hourly build, I'll let you know the result by posting a comment here..

closed 2015-09-17 by ffissore