arduino/Arduino #3282: Compiler hangs on error in include file

nickgammon, 2015-06-04:


nickgammon, 2015-06-04:
A similar issue (with a larger sketch) was reported in the forum at http://forum.arduino.cc/index.php?topic=327636.0

The user there reported using Windows 8.1 (and IDE 1.6.4).

My example above is a small sketch which reproduces the issue.

nickgammon, 2015-06-04:
Compiling under Ubuntu (64-bit) gives a more reasonable error message:

```
Build options changed, rebuilding all
In file included from sketch_jun04a.ino:1:0:
foo.h:1: error: expected unqualified-id before '.' token
 ../foo.h
 ^
```

Plus a whole lot more errors which I am uncertain of the reason for - possibly a side-effect.

ffissore, 2015-06-04:
It's a bug solved in 8a1e6c8a9b328e2aeeac20fa1c5c7db9a825cb18. It showed up as a hung compilation when warning levels were anything but "none". See #3124 and #3115
Fix is available with latest hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-06-04 by ffissore

curie31, 2020-03-25:
> ../foo.h

Hello Sir,
Not able to follow this. Can you please elaborate?

Thanks.

curie31, 2020-03-25:
> Using IDE 1.6.4, Windows XP (64 bit). Build target: Arduino Uno.
> 
> Make a sketch as follows:
> 
> ```
> #include "foo.h"
> 
> void setup() { }
> void loop() { }
> ```
> 
> Make a new tab in the IDE called "foo.h". Place this inside it:
> 
> ```
> ../foo.h
> ```
> 
> The compile process hangs indefinitely with this verbose output:
> 
> ```
> 
> Build options changed, rebuilding all
> 
> C:\Documents and Settings\Nick\Desktop\arduino-1.6.4\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -IC:\Documents and Settings\Nick\Desktop\arduino-1.6.4\hardware\arduino\avr\cores\arduino -IC:\Documents and Settings\Nick\Desktop\arduino-1.6.4\hardware\arduino\avr\variants\standard C:\DOCUME~1\Nick\LOCALS~1\Temp\build1865759335466738571.tmp\sketch_jun04a.cpp -o C:\DOCUME~1\Nick\LOCALS~1\Temp\build1865759335466738571.tmp\sketch_jun04a.cpp.o 
> ```

Hi.
Can you please elaborate this?

Thanks.

per1234, 2020-03-25:
@curie31 this is about a bug that was fixed 5 years ago. There's nothing to elaborate. Hundreds of people get an email every time you comment on one of our issues. That's fine if you have something constructive to add, but that's not the case here.