arduino/Arduino #3286: EducationShield

Alejandro1982, 2015-06-04:
Library used for the completion of the projects of Creative Techonlogies in the Class (CTC)

ffissore, 2015-06-04:
I guess you want it included in the library manager: can you provide the repo url?

Alejandro1982, 2015-06-04:
yes, that is the idea. I just opened the issue in the Arduino Github as
EducationShield and the repo url is:

https://github.com/arduinoverkstad/EducationShield.git
ᐧ

Alejandro Vazquez Bofill
Hardware/Firmware Developer
+46 (0) 760248825
Arduino Verkstad Malmö <http://verkstad.cc/>

ffissore, 2015-06-04:
Added. Thanks!

closed 2015-06-04 by ffissore

q2dg, 2015-06-04:
Would it be possible to add this shield to Arduino official shields (and therefore, to sell it in Arduino store)? It's done by Cuartielles and co...