arduino/Arduino #3293: Hyperlink in comment causes IDE error

per1234, 2015-06-05:
Windows 7 64 bit using 1.6.5 Hourly Build 2015/06/04 08:34
- Type any URL in a sketch
- Inline comment out the URL(`//`)
- The second `/` triggers the error:
```

Exception in thread "AWT-EventQueue-0" java.lang.ArrayIndexOutOfBoundsException: 19
	at org.fife.ui.rsyntaxtextarea.TokenMap.getHashCode(TokenMap.java:202)
	at org.fife.ui.rsyntaxtextarea.TokenMap.get(TokenMap.java:125)
	at processing.app.syntax.PdeKeywords.getTokenType(PdeKeywords.java:178)
	at processing.app.syntax.SketchTokenMaker.addToken(SketchTokenMaker.java:55)
	at org.fife.ui.rsyntaxtextarea.TokenMakerBase.addToken(TokenMakerBase.java:108)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.addToken(CPlusPlusTokenMaker.java:1855)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.addToken(CPlusPlusTokenMaker.java:1839)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.yylex(CPlusPlusTokenMaker.java:2295)
	at org.fife.ui.rsyntaxtextarea.modes.CPlusPlusTokenMaker.getTokenList(CPlusPlusTokenMaker.java:1901)
	at org.fife.ui.rsyntaxtextarea.RSyntaxDocument.getTokenListForLine(RSyntaxDocument.java:416)
	at org.fife.ui.rsyntaxtextarea.SyntaxView.paint(SyntaxView.java:719)
	at javax.swing.plaf.basic.BasicTextUI$RootView.paint(BasicTextUI.java:1434)
	at javax.swing.plaf.basic.BasicTextUI.paintSafely(BasicTextUI.java:737)
	at javax.swing.plaf.basic.BasicTextUI.paint(BasicTextUI.java:881)
	at javax.swing.plaf.basic.BasicTextUI.update(BasicTextUI.java:860)
	at org.fife.ui.rtextarea.RTextAreaBase.paintComponent(RTextAreaBase.java:719)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.paintComponent(RSyntaxTextArea.java:1993)
	at javax.swing.JComponent.paint(JComponent.java:1056)
	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5219)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1572)
	at javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1495)
	at javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:306)
	at javax.swing.RepaintManager.paint(RepaintManager.java:1265)
	at javax.swing.JComponent._paintImmediately(JComponent.java:5167)
	at javax.swing.JComponent.paintImmediately(JComponent.java:4978)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:824)
	at javax.swing.RepaintManager$4.run(RepaintManager.java:807)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:807)
	at javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:782)
	at javax.swing.RepaintManager.prePaintDirtyRegions(RepaintManager.java:731)
	at javax.swing.RepaintManager.access$1300(RepaintManager.java:64)
	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1720)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```
This seems to be caused by moving the horizontal position of the hyperlink text on the screen, you can get the same error by pasting in a commented out URL and then typing any character before the URL.

ffissore, 2015-06-05:
Ahh subtle bug! Thank you!

ffissore, 2015-06-05:
Fixed. Fix will be available in the next hourly build http://www.arduino.cc/en/Main/Software#hourly

commit 0a6841d9 (2015-06-05): New editor: when start is greater than end, delegate to upstream token maker, that will create an empty token. Fixes #3293
closed 2015-06-05 by ffissore