arduino/Arduino #3297: [Boards Manager] Include examples to be added to Examples menu

mikaelpatel, 2015-06-06:
Examples in core libraries are automatically added to the Examples menu. Unfortunately core examples are not. 

Would like to suggest a sub-menu with the core example sketches and libraries instead of a single level. Could be named after the core. 

E.g. 
File>Examples>CORE>Examples>....   the examples directory
File>Examples>CORE>Libraries>...      the libraries directory

where CORE could be, for instance, Cosa. 

Cheers!

ffissore, 2015-06-08:
@mikaelpatel what's a core example? the IDE looks for examples only in libraries folders and in its built-in "examples" folder

mikaelpatel, 2015-06-08:
@ffissore 
Sorry about the somewhat short description. The "issue" occurs when using the new Boards Manager and installing a package/core with it (in my case Cosa). When previously installing to the Sketchbook/hardware/avr folder the Cosa/examples folder was available in the Sketchbook sub-menu. With the Boards Manager only the examples in the libraries folder are available. It would be great to allow a bult-in examples folder per core (same level as libraries) i.e. per package.

per1234, 2017-07-03:
This turns out to be a fairly common need for hardware packages. Some examples:
- https://github.com/esp8266/Arduino/tree/master/libraries/esp8266/examples
- https://github.com/NicoHood/HoodLoader2/tree/dev206/avr/libraries/HoodLoader2/examples
- https://github.com/MCUdude/MegaCore/tree/master/avr/libraries/AVR_examples/examples
- https://github.com/MCUdude/MiniCore/tree/master/avr/libraries/AVR_examples/examples
- https://github.com/MCUdude/MightyCore/tree/master/avr/libraries/AVR_examples/examples
- https://github.com/MCUdude/MicroCore/tree/master/avr/libraries/AVR_examples/examples
- https://github.com/MCUdude/MajorCore/tree/master/avr/libraries/AVR_examples/examples
- https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/A_STM32_Examples

As you see, we found the workaround of using a dummy library for these, which does work.

jpliew, 2018-09-24:
@per1234 ,
The workaround works with 2 confusions

1. If the package has more than 1 board, the board name for the selected board from Tools ->Board will show up on the title of the example.  Take this SparkFun workaround 

[image]

The board selected was SerLCD but the examples was written for LilyPadProtoSnapPlus. Still the examples shows up on File->Examples menu.

2. When user select Sketch->Include Library, the LilyPadProtoSnapPlus showed up as an includable library

[image]

but it is not a library, therefore causing user confusion.

Cheers
JP

jpliew, 2018-09-28:
@per1234 ,

I am sorry if I was not clear with my comment. Causing you to be confused by saying wrong.

> If  the package has more than 1 board, the board name for the selected board from Tools ->Board will show up on the title of the example. Take this SparkFun workaround. 

> The board selected was SerLCD but the examples was written for LilyPadProtoSnapPlus. Still the examples shows up on File->Examples menu.

These two paragraphs were meant to be read as one explanation. When I said

>the board name for the selected board from Tools ->Board will show up on the title of the example.

I was referring to 

`Example for SparkFun SerLCD`

Anyway, if you have not heard of anyone get confused, doesn't mean that no one is confused. The world is so big. Arduino is the most used product at school, anyway we can do to minimized confusion should be **WELCOME.**

Board specific examples is definitely something useful, it may not be for you, but does not mean others will not be beneficial. 

Good luck.

jpliew, 2018-09-28:
@per1234 

> I would have put all the examples under one dummy library:
> 
> ```
> libraries
> |_SparkFun_Examples
>    |_SparkFun_Examples.h
>    |_examples
>       |_LilyPadProtoSnapPlus
>       |  |_LPP_01_Blink
>       |  |  |_LPP_01_Blink.ino
>       |  |_LPP_02_BasicColorMixing
>       |  |  |_LPP_02_BasicColorMixing.ino
>       |  |_etc...
>       |_Qduino
>          |_batteryLeveltoRGB
>          |  |_batteryLeveltoRGB.ino
>          |_fuelGauge
>          |  |_fuelGauge.ino
>          |_etc...      
> ```
> 

Even if someone does the above suggestion, the title of the example will still be called

`Examples for xxxxx`

where `xxxxx` is the selected board. So this suggestion is technically still confusing.