arduino/Arduino #3305: Renaming don't consider case change as a change

simonjamain, 2015-06-08:
I forgot to uppercase the first letter of my class file name, so I go rename via the menu at the right of the tabs. After putting it uppercase I validate but it doesn't do anything. I manage to do it by first rename the file by changing letters and then putting the good file name.

Sorry if bad english.

Simon

ffissore, 2015-06-09:
I think this is a windows issue: windows filesystem is case insensitive so changing a file name from "this" to "This" is not a change. You'd better change "this" to "that" and back to "This"

simonjamain, 2015-06-22:
Thank you for your answer.

But I would like to mention that this "bug" occured on **OSX yosemite** and not on windows, sorry about forgetting to mention that.

ffissore, 2015-06-22:
Ah, that changes. I'll look into it. Thanks

ffissore, 2015-06-22:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-06-22 by ffissore

simonjamain, 2015-06-24:
Many thanks, have a good day.

commit 5fbf9621 (2015-06-22): Sketch rename: allowig a case change rename if NOT on windows. Fixes #3305