arduino/Arduino #3307: IDE quits after splash screen on Windows

rogerclarkmelbourne, 2015-06-09:
Somehow I got the IDE / packages into a position where the IDE just displays the splash screen then crashes and quits

I think this is because the last thing I did was to change the Board manager AVR installed core from 1.6.2 to 1.6.3 and it gave an error.

So I think the steps to repeat this bug would be....

Remove all Arduino installations 
Remove %appdata%\Arduino15 folder

Install IDE 1.6.2

Install Windows IDE 1.6.4 
Select Boards manager
Select AVR and change cores version to 1.6.2

Install will give error message

Quit IDE

Restart IDE (IDE Immediatly quits)

ffissore, 2015-06-10:
I can't reproduce it. Here is what I did: removed %appdata%\Arduino15, started IDE 1.6.4, installed avr core 1.6.2, compiled the default sketch, restarted the IDE
Can you try again starting the IDE from the command prompt using arduino_debug.exe ? It should print an error message

rogerclarkmelbourne, 2015-06-10:
Federico

OK I will try again.

If I'd been clever I would have saved the files in the Arduino15 folder, but its too late now ;-(

rogerclarkmelbourne, 2015-06-10:
I cant re-create it the same way, but I think I can work out how to make it quit in the same way

I noticed a warning about a problem with boards.txt after I'd installed the different version of the AVR core.
But I ignored it

Anyway, if you go into 

packages\arduino\hardware\avr\1.6.3

probably 1.6.7 is the same

and rename boards.txt to boards.txt.old

Then restart the ide it displays the splash screen then quits.

I suspect that somehow the package manager either didn't install a boards.txt of perhaps it had bad data in it.

Sorry I cant tell you how the Boards Manager didn't install boards.txt  but I think this is almost certainly the issue.

ffissore, 2015-06-10:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-06-10 by ffissore

rtek1000, 2019-04-20:
Hi,

Something like that happened to me today.

I was testing the IDE I just compiled with cygwin, but the IDE I compiled failed to open.

So I tried to run the IDE downloaded from the site, but it did not open either.

I found this post, then renamed the Arduino15 folder, to Arduino15.old.

I tried running the IDE downloaded from the site (Arduino) and it opened normally.

So I closed the IDE and went to see the newly created Arduino15 folder.

I copied the contents of the old folder to the new folder, but it does not replace a file with preferences.txt name.

I tried again running the IDE downloaded from the site (Arduino) and it opened normally.

Arduino IDE 1.8.9

It would be great if IDE could open to report the problem with any file. In this way, the user can solve without having to download again the immense contents of the auxiliary folders.

Thank you.

New file (good file):
[preferences.txt](https://github.com/arduino/Arduino/files/3099786/preferences.txt)

Old file (bad file):
[preferences.txt](https://github.com/arduino/Arduino/files/3099787/preferences.txt)

rtek1000, 2019-04-20:
Hi,

Something like that happened to me today.

I was testing the IDE I just compiled with cygwin, but the IDE I compiled failed to open.

So I tried to run the IDE downloaded from the site, but it did not open either.

I found this post, then renamed the Arduino15 folder, to Arduino15.old.

I tried running the IDE downloaded from the site (Arduino) and it opened normally.

So I closed the IDE and went to see the newly created Arduino15 folder.

I copied the contents of the old folder to the new folder, but it does not replace a file with preferences.txt name.

I tried again running the IDE downloaded from the site (Arduino) and it opened normally.

Arduino IDE 1.8.9

It would be great if IDE could open to report the problem with any file. In this way, the user can solve without having to download again the immense contents of the auxiliary folders.

Thank you.

matthijskooijman, 2019-04-23:
@rtek1000, if still have the old broken Arduino15 folder, perhaps you could try putting it back (breaking the IDE again) and run arduino_debug.exe rather than arduino.exe (probably from a command prompt). That should show some more output and perhaps a error message we can work with.

If you do and get some sensible output, it would be best to open up a new issue, rather than continue discussion here. You should probably also include the broken preferences.txt then.

rtek1000, 2019-04-24:
Hi, ok thank you for the attention. I just commented here in case someone else searches for this problem.

commit 0a6577f8 (2015-06-10): If parsing a core fails, don't crash but report a warning. Fixes #3307