arduino/Arduino #3308: Missing line break between compiler warning and multiple libraries notification

per1234, 2015-06-09:
Using Arduino IDE 1.6.5 Hourly Build 2015/06/08 03:34 under Windows 7 64 bit
Very minor issue but I thought I'd mention:
- Include library that has multiple copies installed.
- Add code to sketch that causes a compiler warning.
- Compile sketch - there is no line break between the last warning and the multiple libraries notification.
[image]
If I paste the text here I do get the line break in the proper place.

per1234, 2015-06-26:
This issue is not limited to the multiple libraries notification. For example when 
compiling the Ethernet library AdvancedChatServer example with Compiler warnings: All:
[image]
There is a missing line break between two warnings.

ffissore, 2015-06-26:
I've started working on a fix by reimplementing the EditorConsole. Some initial work is available at https://github.com/ffissore/Arduino/tree/new-console but it's still not mature enouhg

ffissore, 2015-06-29:
Give a try to #3433 it's ready to be merged, it needs just some positive feedback

per1234, 2015-06-30:
Works fine for me. Thanks!

closed 2015-06-30 by per1234