arduino/Arduino #3309: Part of ambiguous overload compiler warning shows with Compiler warnings: None in Preferences

per1234, 2015-06-09:
Using Arduino IDE 1.6.5 Hourly Build 2015/06/08 03:34 under Windows 7 64 bit

Example sketch code to cause ambiguous compiler warning:
```
#include "ambiguous.h"

void setup() {
  B b;
  unsigned int ui;
  foo(&b, ui);
}

void loop() {}

void foo(B* b, signed int si) { } // Overload 1
void foo(B const* b, unsigned int ui) { } // Overload 2
~~~
ambiguous.h
```
struct  B { };
```
Console output with **Compiler warnings: Default** setting in Preferences:
```
ambiguous.ino: In function 'void setup()':
ambiguous.ino:5:15: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [enabled by default]
ambiguous.ino:6:6: note: candidate 1: void foo(const B*, unsigned int)
ambiguous.ino:5:6: note: candidate 2: void foo(B*, int)

Sketch uses 642 bytes (0%) of program storage space. Maximum is 253,952 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 8,183 bytes for local variables. Maximum is 8,192 bytes.
```
Console output with **Compiler warnings: None** setting in Preferences:
```
ambiguous.ino: In function 'void setup()':
ambiguous.ino:6:6: note: candidate 1: void foo(const B*, unsigned int)
ambiguous.ino:5:6: note: candidate 2: void foo(B*, int)

Sketch uses 642 bytes (0%) of program storage space. Maximum is 253,952 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 8,183 bytes for local variables. Maximum is 8,192 bytes.
```

ffissore, 2015-06-09:
Sorry, I didn't catch the problem

per1234, 2015-06-09:
Maybe I am wrong in thinking of the candidates listing as part of the compiler warning but I don't see that this information is useful to someone who doesn't want any warnings displayed.

ffissore, 2015-06-09:
I suppose that's normal gcc behaviour

closed 2015-06-09 by ffissore