arduino/Arduino #3310: pinMode OUTPUT set High By Default on DUE

3esmit, 2015-06-09:
Architecture: SAM
Board: Arduino Due
When the pin is defined as OUTPUT it automatically sets HIGH in that pin. This is different behavior from AVR. 

In Arduino/hardware/arduino/sam/cores/arduino/wiring_digital.c. To make pinMode output LOW by default, change PIO_OUTPUT_1 to PIO_OUTPUT_0.

See http://forum.arduino.cc/index.php?topic=185230.0 and http://forum.arduino.cc/index.php?topic=154869.0

DavidEGrayson, 2015-07-09:
Setting the pin to low by default as pull request #3317 does might be safer than setting it high, but it still doesn't make the behavior of the Arduino Due compatible with other Arduino boards so it's not that great.

On the Arduino Uno and Leonardo, if you want to set an output to be high without having it temporarily drive low (which could break things), then you have to do it with this sequence of commands:

~~~~
digitalWrite(x, HIGH);
pinMode(x, OUTPUT);
~~~~

And if you want to drive it low without temporarily driving it high, you have to use this sequence:

~~~~
digitalWrite(x, LOW);
pinMode(x, OUTPUT);
~~~~

Currently, the first sequence would work on the Arduino Due and the second sequence would not work.  Pull request #3317 simply changes it so that the first sequence doesn't work and the second sequence does work.

In my opinion, for any board to be called "Arduino compatible" *both* of those sequences should work.  If one of them doesn't work, then you will have trouble running code and libraries that depend on the traditional behavior.

There is at least one library where the code had to be changed to work around this non-standard behavior of the Arduino Due: https://github.com/pololu/a4990-motor-shield/blob/master/A4990MotorShield/A4990MotorShield.cpp#L12-24

I haven't looked into the details, but it might be that for the Arduino DUE, the Arduino IDE simply needs to emulate the PORTx and DDRx bits of the AVR.

Even if you don't care about compatibility with existing Arduino code and other Arduino boards, it seems like it is an important feature to be able to set an I/O pin to be low without having it temporarily drive high, and vice versa.  Right now it looks like the Due just doesn't support that.

3esmit, 2015-07-10:
I see that my issue is wrongly described, sorry about that.
I forgot that my code did a loop in all digital ports setting them to LOW at setup(), and my pinMode was called after on demand by a protocol message request. 

If I do not call digitalWrite before pinMode, the default Arduino output would be HIGH?

The workaround in A4990MotorShiled works great? I mean, it never gets high, not even by a millisecond? 

The issue in Due will ever be fixed? 
If not, a great feature would be function overloading of pinMode, having a void pinMode(PORT, DIRECTION, CONFIG), where in case of output direction, config HIGH/LOW would set the initial output, and in case of input, config HIGH/LOW would set the pullup feature.

DavidEGrayson, 2015-07-10:
The workaround in the A4990 library is not perfect, no.  The line will be high for a short time after we call pinMode on it.

3esmit, 2015-07-10:
@DavidEGrayson Can you set the labels? If there is another open issue, please mark this as duplicate and close it. Thank you.

DavidEGrayson, 2015-07-10:
Sorry, I'm not an Arduino developer, just someone commenting on this issue.

cmaglie, 2015-07-15:
Would you like to try the solution on #3524 and give feedback there?

facchinm, 2016-01-07:
Solved by #3524

closed 2016-01-07 by facchinm