arduino/Arduino #3311: Arduino IDE serial port selection unavailable

Avamander, 2015-06-09:
When my copy of Arduino IDE (2015/06/05 7:26)  is started when an Arduino is connected on Ubuntu 15.04 64bit edition computer I can not select any of my Arduinos. When I unplug them all, it starts fine and I am able to select Arduinos I plugged in after the startup. Giving Arduino IDE root permissions did not change anything.

Picked up JAVA_TOOL_OPTIONS: 
Exception in thread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" java.lang.NullPointerException
	at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:143)
	at processing.app.linux.Platform.resolveDeviceAttachedTo(Platform.java:144)
	at cc.arduino.packages.discoverers.serial.SerialBoardsLister.run(SerialBoardsLister.java:83)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)

ffissore, 2015-06-09:
Are you using the hourly build?

ffissore, 2015-06-09:
This issue seems a duplicate of #2892 which was fixed with 1.6.4

zerog2k, 2015-06-11:
I'm having the same issue all of the sudden with 1.6.4.
Can't be for sure, but it might have something to do with vmware player. I can see the ports /dev/ttyUSB*, but even vmware cant see the ports to assign them to the vms.
When I suspended the vm and restart arduino in ubuntu 14.10 it now works again.

fosdex, 2015-06-12:
I can confirm this problem with arduino-1.6.4 on openSUSE 13.2. In two specific use cases the Arduino IDE generates a JAVA runtime error and the Tools->Port menu item is greyed out so the board is not selectable or loadable.

This problem always occurs in two specific use cases and does not matter whether running as root or user:

1) The first time run of the Arduino IDE after installation when an Arduino board is plugged into a USB port before starting the IDE, and,
2) The first time run of the Arduino IDE after login when an Arduino board is plugged into a USB port before starting the IDE.

The key to the problem is an Arduino board being plugged into a USB port before starting the Arduino IDE and the instance of the Arduino IDE being the first instance (occurance) of a login session. If no board is plugged into a USB port under the two use cases above then no JAVA runtime error is generated and the menu item Tools->Port is not greyed out, that is, the IDE is fully functional.

After the first time run of the Arduino IDE, with or without an Arduino board plugged in before starting the IDE, the IDE will subsequently run with an Arduino board plugged in before starting the IDE without problems.

The workaround is to not plug an Arduino board into a USB port until after the first instance Arduino IDE is running in a login session.

This problem is reproducible always.

Example of a first time Arduino IDE run in a login session:

cosmic@linux-7q19:~> java -version
openjdk version "1.8.0_45"
OpenJDK Runtime Environment (build 1.8.0_45-b14)
OpenJDK 64-Bit Server VM (build 25.45-b02, mixed mode)
cosmic@linux-7q19:~> lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 011 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 009 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 004: ID 03f0:8811 Hewlett-Packard 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 004: ID 0764:0501 Cyber Power System, Inc. CP1500 AVR UPS
Bus 005 Device 003: ID 045e:00cb Microsoft Corp. Basic Optical Mouse v2.0
Bus 005 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120 for Business
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 002: ID 2341:0042 Arduino SA Mega 2560 R3 (CDC ACM)
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
cosmic@linux-7q19:~> ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Jun 11 22:05 /dev/ttyACM0
cosmic@linux-7q19:~> id cosmic
uid=1000(cosmic) gid=100(users) groups=14(uucp),16(dialout),54(lock),100(users)
cosmic@linux-7q19:~> arduino
Exception in thread "cc.arduino.packages.discoverers.serial.SerialBoardsLister" java.lang.NullPointerException
	at processing.app.Platform.resolveDeviceByVendorIdProductId(Platform.java:143)
	at processing.app.linux.Platform.resolveDeviceAttachedTo(Platform.java:142)
	at cc.arduino.packages.discoverers.serial.SerialBoardsLister.run(SerialBoardsLister.java:83)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)

At this point the Tools->Port menu item is greyed out and the board is not selectable or loadable.

Leaving the Arduino board plugged into the USB port and quitting then restarting the Arduino IDE results in no JAVA runtime error and the Tools->Port menu item is not greyed out - the IDE is now fully functional.

My system configuration is stock openSUSE 13.2 GNOME 64-bit with OpenJDK 1.8.

cosmic@linux-7q19:~> uname -a
Linux linux-7q19 3.16.6-2-desktop #1 SMP PREEMPT Mon Oct 20 13:47:22 UTC 2014 (feb42ea) x86_64 x86_64 x86_64 GNU/Linux

The user is configured into the uucp, dialout, and lock groups.

I do not run vmware so this is not the problem as was speculated in another post. If this issue is a duplicate of #2892 then it was NOT fixed with 1.6.4.

ffissore, 2015-06-12:
I think I fixed it. It could have been due to your computer taking more than 5 seconds to start parsing the hardware. Now SerialBoardsLister waits indefinitely and it shouldn't fail any more. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-06-12 by ffissore

Avamander, 2015-06-16:
2015/06/16  04:19 build does not have the issue. It indeed seems to be fixed, many thanks!

ffissore, 2015-06-16:
Good to know! Fix is part of IDE 1.6.5, released yesterday

commit a755dd76 (2015-06-12): SerialBoardsLister now waits indefinitely before starting. BaseNoGui.packages
will eventually be set to at least an empty HashMap, even if no hardware is
present. Fixes #3311