arduino/Arduino #3313: 3rd party Hardware specification for ST plugin

gepd, 2015-06-10:
I'm trying to fix an issue with the plugin stino (https://github.com/gepd/Stino) for Sublime Text who has some problem with versions 1.6.x

it compiles without problems the script, but when I try to upload it to the board, it shows the error:

`"{runtime.tools.avrdude.path}/bin/avrdude" "-C{runtime.tools.avrdude.path}/etc/avrdude.conf" -v -patmega328p -carduino "-PCOM1" -b115200 -D "-Uflash:w:C:\arduino-1.6.4\build\test.hex:i"`

I modify the `platform_wiring.txt` and `platform_avr.txt` (in the plugin) based on the `platform.txt` file located in `arduino-1.6.4\hardware\arduino\avr`

but it didn't fix the problem, can anyone tell why can be this happening?
what are the changes introduces in version 1.6.x

plus, there is not 1.6 hardware specificacion guide like this https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

ffissore, 2015-06-10:
Hardware specification has not changed since 1.5.
As for the missing `{runtime...}` variables, they are available running the latest hourly build of the IDE (soon to become 1.6.5) with the command `./arduino --board PACKAGE:ARCH:BOARD_ID --get-pref`: this will dump all the preferences. As of now, you should run this command every time your users select a different board

gepd, 2015-06-10:
Thanks for your answer ffissore,
I've tested the new hourly build and do some test, but the error continues

I forgot to write some part of the error:

`"{runtime.tools.avrdude.path}/bin/avrdude" "-C{runtime.tools.avrdude.path}/etc/avrdude.conf" -v -patmega328p -carduino "-PCOM1" -b115200 -D "-Uflash:w:C:\arduino-1.6.4\build\test.hex:i"
The system cannot find the path specified.`

I have a doubt, what should be the value of `runtime.tools.avrdude.path` ? is that kind of cariables documentated? same as `runtime.tools.avr-gcc.path`

gepd, 2015-06-10:
Never mind, I fixed it is a variable that I have to define in the plugin.
Thanks for your suppor, you can close this issue now.

closed 2015-06-15 by gepd