arduino/Arduino #3318: RAMBo motherboard missing

a4jp-com, 2015-06-11:
I moved the files for the RAMBo motherboard into the hardware folder but the RAMBo motherboard option doesn't come up.

Without this my Printer can't load the LCD driver.

Please help.

Windows 7 64-bit

ffissore, 2015-06-11:
Please ask your question to RAMBo support/issue tracker/forum

closed 2015-06-11 by ffissore

a4jp-com, 2015-06-11:
Please add RAMBo boards to Arduino.

a4jp-com, 2015-06-11:
From version 1.0.5 to 1.6.4 what changed to make the RAMBo board not load? I guess the change was programming here.

ffissore, 2015-06-11:
Indeed hardware folder structure changed. See this guide https://github.com/arduino/Arduino/wiki/Arduino-Hardware-Cores-migration-guide-from-1.0-to-1.6 Please forward it to RAMBo developers