arduino/Arduino #3319: chain Strings and chars in a String

RobertoMaiocco, 2015-06-11:
hi, maybe I'm wrong but i cannot understand the reason for a compilation error i got by compiling the following test program (i got the same error in a "real" program)

char ht,cnd,lg,al;
String parametri;

void setup() {
  // put your setup code here, to run once:
ht = 'Y';
cnd = 'Y';
lg = 'Y';
al = 'Y';

// THIS COMPILES
//    parametri = "heat="+String(ht)+"&cond="+cnd+"&light="+lg+"&alarm="+al;

// THIS DOESN'T COMPILE
    parametri = "heat="+ht+"&cond="+cnd+"&light="+lg+"&alarm="+al;

// EVEN THIS DOESN'T COMPILE
  parametri = String("heat="+ht+"&cond="+cnd+"&light="+lg+"&alarm="+al);

}

void loop() {
  // put your main code here, to run repeatedly:

}

THE COMPILER SAYS:

testString.ino: In function 'void setup()':
testString.ino:11:28: error: invalid operands of types 'const char*' and 'const char [7]' to binary 'operator+'
invalid operands of types 'const char*' and 'const char [7]' to binary 'operator+'

cmaglie, 2015-06-11:
This is a duplicate of #1319 (where you can find an explanation of the issue).

Briefly, this is expected and it depends on how the C++ language is defined, we can't do much to solve it.

If you have further programming questions please ask on the forum: https://forum.arudino.cc/

thanks!

closed 2015-06-11 by cmaglie

Chris--A, 2015-06-11:
@RobertoMaiocco 

And as per the examples provided:

> // THIS COMPILES
// parametri = "heat="+String(ht)+"&cond="+cnd+"&light="+lg+"&alarm="+al;

This actually a bonus, it compiles because a specific global operator has been defined for the case of a c-string + String. Then the rest of the statement contains expressions with a String on the LHS and work as normal.

https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/WString.h#L125

For the other two examples, make the first item of the chain of a+b+c a String object:

```Arduino
parametri = String("heat=")+ht+"&cond="+cnd+"&light="+lg+"&alarm="+al;

parametri = String( String("heat=")+ht+"&cond="+cnd+"&light="+lg+"&alarm="+al);
```

RobertoMaiocco, 2015-06-11:
Ok tks I'll look for issue 1319.

Roberto