arduino/Arduino #3321: ArduinoISP: reliability and portability improvements.

PeterVH, 2015-06-11:
I recently picked up the effort to make ArduinoISP more reliable and better working on more (hopefully all) arduino models. I started again from rsbohn's last work and try to fix only things that help achieving these two goals.

This effort is in my [master branch] (https://github.com/PeterVH/ArduinoISP/tree/master).

**Status**

This is still a work in progress, but here is a summary of the status:
* Rsbohn's last work introduced some regressions, but these are fixed.

* On the Uno it works much more reliable than the version in the IDE, mainly by simply removing the delay() call from the heartbeat() method.

* On the Due. There is no bit bang spi yet: I think this would better be handled by a SPI.setSpeed() method in the SPI lib (if the selected speed is too slow, the implementation could automatically switch to bit banged SPI).

Therefore on the Due it is already working (untill now I only tested reading signatures) but not for slow targets like the attiny85...

**Testing:** I am composing a test suite bundling a set of about 10 typical/important use cases for ArduinoISP. The suite needs to be fleshed out, but here is a preview:
1. No target connected.  (note this does not work as it should with the current version in the IDE)
2. Burn UNO bootloader
3. Burn mega bootloader (requires programming above 128KB)
4. Program an attiny85
5. Program an attiny841
6. Program an at89s52
7. Chaucer115 to 1284p: Program a big (115KB) sketch into an atmega 1284p
8. Chaucer115 to mega: Program a big (115KB) sketch into an arduino mega
9. LongStoryShort: Program above 128KB.

(the name Chaucer comes from the optiboot test suite that burns big sketches that output part of the "knight's tale" by Chaucer on the console)

**Avrdude:** I think it would be better to upgrade to avrdude 6.1 because this version implements some clever fixes that make it possible to program high addresses (over 128KB) with stk500v1 (which is what ArduinoISP uses). This makes it possible to reliably burn the bootloader of an atmega2560 using ArduinoISP.

facchinm, 2015-07-10:
Hi @PeterVH , I'm keeping an eye on your repo and it seems that the Bitbang SPI on Due is solved :metal:
If you could open a proper PR the bot will automatically create a build so people could start testing it easily!

PeterVH, 2015-07-10:
Hi @facchinm, thanks for giving it a try. Beware, I did a push -f a couple of times last days. Of coarse I will stop doing that from now on.

Just created the pull request. (Why did this create a new issue?, could I have avoided that?)

On linux I carried out the test "4. program an attiny85", on a Leonardo, Due and Mega 2560. 
It should work on a Zero too, but I can't buy one from Belgium. 

I elaborated the test suite, and created some more background info about this enhancement request. But it is not yet ready to publish (say mid next week).

TODO: 
- Fix small issue: error led pulses upon every run (because pmode is entered twice).
- I should adapt the config such that on a gallileo it works too (using bit bang). Have no gallileo though.
- Further testing (also windows. And find a mac user that can do some tests on mac).
- see what we can do for  the "-c arduino" thing for windows users.
- create awareness of the new "universal wiring" (always use pin10, use SPI from SPI header, optionally power target from IORef pin))

nerdralph, 2015-08-10:
There's more than one fork to fix/improve ArduinoISP (but too many forks spoil the code...)  ArduinoISP included with 1.6.3 works OK with a target connected, but without a target connected it hangs at the programming command (heartbeat stops cycling):
avrdude -C /etc/avrdude.conf -c avrisp -p t85 -P com16 -b 19200 -vvvv

avrdude: Send: E [45] . [05] . [04] . [d7] . [a0] . [01]   [20]
avrdude: Recv: . [14]
avrdude: Recv: . [10]
avrdude: Send: P [50]   [20]
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding

I tried Randall Bohn's version and it gracefully handles the case of no target connected:
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.

I imagine Peter's version also works OK since it is forked from Randall's.

closed 2015-10-20 by facchinm