arduino/Arduino #3331: Can not start IDE by double clicking on .ino after update to nightly-windows_2015-06-12

bgma, 2015-06-13:
Hello,
After updating the IDE to the Arduino-1.6.5-nightly-windows_2015-06-12 from Arduino-1.6.5-nightly-windows_2015-05-20 can not I start by double clicking on .ino / .pde longer.
I run on W7 pro sp1.
The problem disappeared when I backed off to 2015-05-20 and came back when I updated to 2015-06-12 again.
/Sincerely Bengt-Göran

ffissore, 2015-06-15:
We removed windows registry manipulation code in f6546aea4527bb0723fac6472c1e10328ebe9473 (committed 20th of May) because windows installer is already taking care of it. Therefore, when using the nightly, .ino file will not get associated to the latest run IDE. Thanks for the heads up!

bgma, 2015-06-16:
The problem is still there ...

ffissore, 2015-06-16:
Confirm. This is fixed with dd5edd1b7f6f2df7d83469e8e6186c82dcfa8798 but we are going to release a windows 1.6.5-r2 version to fix this specific issue. Thank you for your patience

bgma, 2015-06-18:
Hello,
just want to confirm that it is working double-clicking .ino file in version nightly ARDUINO 1.6.5-r2 - 17.06.2015 which says Version 1.6.6 Hourly Build 18/06/2015 09:41 (does not matter)

/ Regards Bengt-Göran

ffissore, 2015-06-18:
1.6.5-r2 from the download page says "1.6.5" as expected. If you read "1.6.6 hourly ..." you're using the hourly build

bgma, 2015-06-18:
Yes that's right. It was I who had some version of confusion.
Thanks for the help!
/ Sincerely BG

ffissore, 2015-06-18:
Thank you for the heads up ;)

closed 2015-06-18 by ffissore