arduino/Arduino #3332: Library mananger submenus cosmetic issue - observed on Due.

VaclavSal, 2015-06-13:
Current 1.6.4. release Library Manager lists / display submenu items ("Arduino/SAM libraries") ( after separator ) grayed out. This menu item can be selected ( full blue color) but it does nothing. 
If intended to be used as a title of the following submenus it should be standard (full) text color with ellipsis followed with real submenu list.

ffissore, 2015-06-13:
Can you attach a screenshot of an app that behaves the way you describe?

VaclavSal, 2015-06-13:
A menu item NOT selected 
B menu item SELECTED

[image] 
[image]

Chris--A, 2015-06-14:
But can you actually select it, or does it do nothing.

On Windows 8.1 it looks like this, and it is not clickable:
[image]

VaclavSal, 2015-06-14:
Yes it does nothing and that is why I brought it up as a "cosmetic" issue.
It is an unusual usage of "Menu Control" independent ( or should be) of OS or  software used to build it.
Ideally the "Include Library" menu should look this way

Manage Libraries...    (or "Manage Libraries > "  to be consistent with the parent menu title ) 
Separator
Add Zip Library..
Separator
Arduino Libraries ( Or Adruino /SAM libraries..)  - opens submenu 
Separator 
and so on

That would be consistent usage of menu control, however , an average " push button and light up LED" user probably does not care - hence "cosmetic issue".
.

Chris--A, 2015-06-15:
I agree, that it would be handier to use sub menus rather than a huge list of boards from every core.

ffissore, 2015-06-16:
Submenus VS. disabled menu entry have been debated in the past but are a Processing inheritance we prefer to stick to. Closing as wontfix

closed 2015-06-16 by ffissore