arduino/Arduino #3333: Wrap Around for Higher (>10) Serial Port Number in Windows 7

xsk8rat, 2015-06-13:
When the serial port for the Yun is larger than "Com10", (shown as COM20 in the example image):
[image]
The Tools menu of the Arduino App reports that both the N and Int[N/10] ports are occupied by the Yun:
[image]
Of course, only one is really there.

ffissore, 2015-06-17:
Next time it happens, can you open a command prompt, `cd` to `IDE_FOLDER\hardware\tools\`, run program `listComPorts.exe` and paste the output? That will help understand where the problem is

xsk8rat, 2015-06-18:
Thanks! Here is the output you requested. Please note that i manually changed the port to the higher COM12 using the Device Manager.

    COM1 - (Standard port types) - ACPI\PNP0501\1
    COM2 - (Standard port types) - ACPI\PNP0501\2
    COM12 - Arduino LLC (www.arduino.cc) - USB\VID_2341&PID_8041&MI_00\8&AB76839&0&0000
    COM3 - FTDI - FTDIBUS\VID_0403+PID_6015+DA00WSEWA\0000

ffissore, 2015-06-18:
Ah, weird. I'll test it. Does COM1 stays labelled as "yun" even after some minutes?

ffissore, 2015-06-18:
That was indeed a bug in the parser. Thank you very much. Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-06-18 by ffissore

xsk8rat, 2015-06-18:
Cool! Thank you!

commit 8c60054a (2015-06-18): Windows: ListComPortsParser was reporting the same VID/PID for both COM12 and COM1. Fixes #3333