arduino/Arduino #3334: Adding ARM builds for Raspberry Pi and co.

CRImier, 2015-06-13:
Hello!

I've managed to compile and run Arduino IDE 1.6.5 on a Raspberry Pi (2B), running Raspbian Jessie. I have added a new platform to build.xml, as well as replaced libastylej.so with a copy from Raspbian repos and all the avr-gcc&avrdude binaries by symlinks to the corresponding binaries on my system. It runs quite nice, I've uploaded a Blink sketch on an Arduino Pro Mini. 

Is somebody interested in completing that so that Arduino could offer pre-built IDE archives for Raspberry Pi&ARM? Basically, all that's left is to package libastylej.so, avr-gcc and avrdude into tarballs, as it is already with Linux IDE archives, upload those archives on the official page (where the build script downloads them from) and finish the build.xml work. Then I guess even Raspbian repos could have it, as it'll be officially supported - and, honestly, being stuck on 1.0.5 sucks a lot. 

I have the build.xml file in my copy of this repository ([commit](https://github.com/CRImier/Arduino/commit/6f1bfe1057d80146f423ff357a1f9e5c42613f6e)). I'm not making a pull request because it is far from complete, given that it copies libastylej.so from the system-wide libraries and does nothing about absence of avr-gcc and avrdude packages on Arduino website (I symlinked the binaries to system-wide for IDE to work). I've also run 'ant test' and it did well - only some tests related to autoformat feature failed, overall it was okay.

CRImier, 2015-06-13:
I've also [described the build process](https://crimier.wordpress.com/2015/06/13/compiling-the-latest-arduino-ide-for-raspberry-pi/) on my blog, if that helps.

NicoHood, 2015-06-14:
I tried your tutorial.
I installed all needed packages except avr-g++. Couldnt find the package for some reason.
ant worked anyways and ant run as well. Starting directly from work/linux also works.

It just wont compile. And I also havent added the symlinks which might be the problem. It would be cool if you add the few commands which we need to link those files. It would be cool to already add this in the build process? Or maybe you could add it to the install script (also have a look at the current PR about the install script!).

So it might be just the symlinks for me. the eroor was a syntax error in avr-g++ by the way. The rest was super easy and it would be really cool to see this in the official repository from arduino github and the jessie repo of the raspi debian for pi2.

CRImier, 2015-06-14:
I have met that syntax error with avr-g++ - something about symbol '(', right? That's what it says when version of avr-g++ is for x86, not ARM, even though that's certainly not the most correct error message ever. As far as I understood, you still do have x86 avr-g++, so this error makes sense. 
Do you have 'gcc-avr' package installed? When it's installed, the 'avr-g++' binary can be found in /usr/bin. About symlinks - I'll go and make a script now.

CRImier, 2015-06-14:
So here's an one-liner:

    for i in *; do echo $i; rm $i; ln -s /usr/bin/$i $i; done

Worked for me, will add this to my blog entry now as well.

ffissore, 2015-06-15:
In order to have a ARM compatible version, we need to pack everything we pack now, including astyle, all the toolchains (gcc, avrdude, bossac, openocd)
You may want to start from astyle repo https://github.com/arduino/astyle/ (it contains the scripts we use to build it)
You can then move to https://github.com/arduino/toolchain-avr : we already have an ARM script, but we have not tested it thoroughly.
And in case you'd ask "why don't you just use gcc provided by debian?" the answer is "every version of the IDE (win, mac, lin) must be equal to one another: relying on software we don't ship makes support harder"

CRImier, 2015-06-15:
In this specific case, is there any kind of cross-compiling environment needed when compiling libastyle and binaries for ARM? I figure that your build environment is x86/x64, and I haven't yet compiled ARM things on x86, so I'm kinda confused =)

ffissore, 2015-06-15:
we don't cross compile

NicoHood, 2015-06-15:
Yes I got that symbol '(' error. Yes I've installed gcc-avr. But where to move which file? If you are doing a script then you may want to add the copy command for avr-g++ as well. Do I have to do this before the build or afterwards?

This is the PR I am talking about. You could add your script stuff here. Even though its just a trick right now, many people would appreciate it. https://github.com/arduino/Arduino/pull/3226

ffissore is right about shipping a copy with the ide, so it works under every OS. I tried with a fresh noobs booted debian jessie (upgraded before). I am not sure if wheezy would work since avr-gcc 4.8.1 is only available on jessie.

But if we can fix those bugs we can probably see the IDE also running on wheezy.

CRImier, 2015-06-15:
@NicoHood - Aha, so IMO Wheezy is out of question until we get all the things packaged, or alternatively those who need it can use packages from jessie (and hope it doesn't break things). I think I'll work on compiling&packaging things then.

You don't actually need to move files. All you need is to create symlinks using that one-liner - it deletes all the binaries (extracted from an archive meant for x86, so useless to us) and creates symlinks to according binaries in /usr/bin instead.

NicoHood, 2015-06-15:
I tried this with 1.6 RC1 once. You can set jessie as 2nd repository. But this will break weird things. You cannot download some normal wheezy packages etc. So I wouldnt do it again like this. shipping it with the ide is probably the best thing you could do.

Is this version fixed for pi2 or does it also work on the pi1? Or in other words: Can we improve the performence for pi2 maybe since it has 4 cores? I couldnt compile so I dont know if it uses all 4 cores at the same time. The underlying wheezy/jessie is still ARM6 from what I understood so we cant get ARM7 optimization? But what about the ubuntu mate release that's out now? Would the IDE work there as well?

CRImier, 2015-06-15:
@NicoHood  - I have only tested it on Pi2 so far, but I can also test it on Pi1B+ with Raspbian Jessie, which I'll do right now. I wonder if it'll break =) You're right about the whole thing still being ARMv6 - Raspbian repositories didn't get further than that AFAIK. Also, multicore support should be as good as Arduino IDE itself supports it, that is, if it does set '-j{cores}' flag when calling compiler/linker.
I figure the IDE will work on Ubuntu Mate, since it's shouldn't be that far from Jessie. However, this is nothing more than a wild guess and I'll test it as soon as I get a possibility to finally try it out.

ShorTie8, 2015-06-15:
Other then the IDE there is no need to compile anything because it is all in apt-get.
Why recreate the wheel when the wheel spins fine ??

Have A Great Day
ShorTie

CRImier, 2015-06-15:
@ShorTie8 
The reason has been mentioned by ffissore - it's ease of technical support, and I personally agree with this, especially given that it's basically a platform for Linux beginners. They do the same for Linux i*86 as well. So far this approach can make it work for various distributions, regardless of a package manager. If it, as claimed, helps the maintainers make sure all components are interacting with each other well, I don't see a problem - just an advantage. 
Also, if they were to rely on repositories, they'd need somebody to be package maintainers for all the distributions they want to support (this is open source, after all) - and as for now they support most of the distributions just fine, just not using the preferred way. I'd want to be a maintainer, at least for Debian and Ubuntu, but as for now I'm quite busy and can only do so much.
So why make the wheel more complex when it's easier that way and spins fine?

CRImier, 2015-06-19:
Hi all!
So I've finally compiled the libastylej&toolkit-avr. Looking forward to testing them with  RPi (need to upload them somewhere&&get a direct link to do this), then I'll post them here. My question is - what's the tool for creating checksums for files?

ffissore, 2015-06-19:
Shasum

NicoHood, 2015-06-19:
This looks weird if you delete a post. More information would be great.

ffissore, 2015-06-19:
Indeed. I was answering this question
Arsenijs ha scritto il 19/06/2015 alle 07:02:

PaulStoffregen, 2015-06-19:
I recently built the ARM toolchain for Raspberry Pi.  If this thing is really going to happen, I'd be happy to help with that part.

Many users have asked me to support Raspberry Pi.  The problem has always been a lack of an official Arduino release.  The distro packages are poorly made and tend to have lots of problems, far too many to try supporting them.

CRImier, 2015-06-20:
Hi! 
@PaulStoffregen  I'm now working on making it happen =) Right now am running 'ant' with new toolchain files (had to change 'download.arduino.cc' to 'localhost' and serve files on HTTP, would've been too bothersome to change the script IMO) I'll certainly need help if things go wrong, and they might as well. My first question is - will a toolchain build made on Raspberry Pi 2 (Raspbian Jessie)  (without any compiler flags that'd change architecture) work on Raspberry Pi 1? I'm about to figure that out myself but if you could just clarify that, that'd have been awesome. The best would be explaining the basic principles of RPi1/RPi2 code compatibility and how to compile things on RPi2 to make them work on RPi1=)
Also, there's another really important question. How do we integrate ARM builds into Arduino build process? Do we compile on an ARM device or set up a cross-compilation environment? What needs to be done, and how can I help with that?
How do I extract the toolchain for testing if I have an already-compiled version of IDE, let's say, in /usr/share/arduino? I'll need to test the toolchain multiple times maybe, so a path where to extract files would be nice, to avoid running 'ant' multiple times.

Also, @ffissore I'm sorry for that deleted comment. Just got confused.

CRImier, 2015-06-20:
Hey! I've compiled and packaged a version of IDE, now testing it. It works on Raspbian Jessie on RPi2 (the same machine I used for compiling it). It stops on launch on Raspbian Wheezy on RPi1B, however. The error message is as follows:

```/home/pi/arduino-1.6.5/lib/libastylej.so: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version GLIBCXX_3.4.20' not found (required by /home/pi/arduino-1.6.5/lib/libastylej.so)```

Does that mean that Wheezy simply doesn't have packages recent enough to support the IDE? Or is there a workaround I yet don't have a clue of, like compiling for an older GLIBCXX version? 
As soon as I get to testing it with Jessie on Raspberry Pi 1B or 1B+, I'll let you know about the results. 
Meanwhile, here's [the package](https://www.dropbox.com/s/14v7ill6u68y6dd/arduino-1.6.5-arm.tar.xz?dl=0) .

cmaglie, 2015-06-20:
The error may means that you do not have the stdc++ library installed or that it is older than <3.4.20.

If the library is not installed a `sudo apt-get install libstdc++6` should solve the problem.

If it's an older version the simpler solution is to build `astyle` (or any other native code) directly in the environment that contains the older library (the idea is that new libraries should be compatible, so if you manage to compile for an older release it should be fine with the latest).

CRImier, 2015-06-20:
What I got to know using readelf: compilers in Raspbian produce armv6 code, even if running on armv7. That's great to know. I also more or less understood about the toolchain extraction, and hope I won't really need that (seems I won't need it that much though). Now I also sorted it all out about the basics of compatibility and such =)

@cmaglie libstdc++6 is there indeed. Though the host uses libstdc++-4.9-dev package (without 6 after ++), and target uses libstdc++6-4.6-dev (metapackage is installed on both of them, MB the problem will go away after dist-upgrade I'm doing now on target).

What'd be the more complicated solution? We need to build astyle + **avr toolchain**, and the latter is heavy (essentially building a compiler). That'd be taking a Wheezy image (it has the library that's compatible, I think), and that's something I can't do now, as compilation takes ages, and that RPi 2 I can compile on needs Jessie on it, since it's my desktop PC for now. Maybe there's a possibility to compile against earlier versions of this library? There should be, I just don't know how yet =(

cmaglie, 2015-06-20:
> Maybe there's a possibility to compile against earlier versions of this library? There should be, I just don't know how yet =(

From my experience I can tell you that is so complex and error prone that many prefers to create a virtual-machine with the same setup as the target. BTW even the VM may be slow as hell, with the drawback that you waste a lot more energy to run the VM on a PC compared to a Raspberry.

CRImier, 2015-06-20:
Hmm. @cmaglie  - would your approach work if I set up a Wheezy x86 machine and cross-compile?

Honestly, can't wait until the Foundation-provided image becomes Raspbian Jessie.

CRImier, 2015-06-20:
Sorry, wrong button.

ShorTie8, 2015-06-20:
libastylej is not in wheezy, that is why you need jessie
https://packages.debian.org/search?keywords=libastylej-jni

Have A Great Day
ShorTie

@PaulStoffregen about as official as I can seem to make it, but you still need jessie
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=92662

CRImier, 2015-06-20:
I'm currently setting up a Wheezy compile environment on a Raspberry Pi 2 to compile that libastylej and the toolchain too. If I don't stumble upon any problems, I'll have the whole system working on Wheezy. I hope nothing has any requirements that can't be supported on Wheezy.

@ShorTie8 I'm sure patching is not the official way of doing things, as it's IMO hard to maintain. And I hope they get to Jessie soon, but I have to admit Jessie has some serious web browser issues as for now. (offtopic: if you have the same webbrowser problem, try luakit.)

PaulStoffregen, 2015-06-20:
@ShorTie8 In my mind, the only "official" release is if Arduino.cc publishes it as a stand-alone download.  The  distro packages tend to be a support nightmare, especially Debian!

CRImier, 2015-06-20:
Got it working on Wheezy! Recompile in Wheezy environment did the trick, now recompiling the toolchain-avr since it misbehaves too, in the same way as astyle did, just complaining about GLIBC_2.17 not found. Right now IDE runs, just doesn't compile =) 

@ffissore @PaulStoffregen So far, it doesn't seem to have unsolvable problems (toolchain's still compiling but I hope for the best ;-) ) When I get it working on Wheezy, what do I do next? Do I send you my build environment? (I could package the quick Wheezy setup I just did, just 4GB, runs on RPi2). Which files do I need to provide? What's about testing? Anything else that'd be desired?

CRImier, 2015-06-21:
All working and [packaged](https://www.dropbox.com/s/m4pp9stjhsrc0nf/arduino-1.6.5-arm_wheezy.tar.xz?dl=0). Moreover, the package runs both on Wheezy and Jessie without any problems. Waiting for further directions =)

ShorTie8, 2015-06-21:
Whelp, for yucks I tried it. Got a few warnings, but did seem to compile sumfin.

Picked up JAVA_TOOL_OPTIONS: 

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-entry-select-on-focus after class was initialised

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-entry-password-hint-timeout after class was initialised

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-can-change-accels after class was initialised

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popup-delay after class was initialised

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popdown-delay after class was initialised

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-bar-popup-delay after class was initialised

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-label-select-on-focus after class was initialised

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-scrolled-window-placement after class was initialised

(java:5333): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-button-images after class was initialised

ShorTie

CRImier, 2015-06-22:
@ShorTie8 Glad it worked for you! Those warnings don't mean much, they're due to the way how some of the UI elements are coded in the Arduino code =) I think I haven't seem them in the version that was compiled on Jessie, obviously the GTK library has changed and doesn't throw these warnings anymore.
Did you try it on Jessie or Wheezy? Would be better if you tried Wheezy - I want to know how it works on other setups. Also, did you try to upload something?

ShorTie8, 2015-06-22:
I tried it quickly on wheezy to see if it ran and would compile something.
No, I did not try to upload anything.
That is about all the testing for me, like I have said, I normally don't use 3rd party binaries because you never know what is in them, Sorry.

ShorTie

VogonJeltz, 2015-06-28:
Arduino IDE 1.6.x on Raspi sounds great! Does it work meanwhile or not?

CRImier, 2015-06-28:
Hi!
What you're talking is awesome but is not exactly what is going on here =) It's all about making the IDE run on Raspberry Pi, not adapting Arduino toolkit to make it support all Raspberry Pi features as a devboard. That means - just running IDE and enabling it to compile/upload code to various Arduino boards now supported by Arduino IDE.
What you're talking about would be WiringPi, though it does not exactly support HDMI or Audio, but it's what you described. I'd also suggest Python - it's hella easy to code and supports all of the mentioned features as of now =)

CRImier, 2015-06-28:
@ffissore @PaulStoffregen I think I'll prepare a pull request with additional instructions and let you guys see what you can do with it =)

Nulldev128, 2015-08-02:
Hi CRImier!

I have an Arduino Due hooked up to my Raspberry Pi 2 hosting emoncms for energy monitoring.
I've followed your instructions & got the IDE installed & working perfectly on wheezy using the archive your linked above via dropbox. Thanks very much for all your hard work!

When I go to verify or upload the sketch, I get the following output:
<hr>
Arduino: 1.6.5 (Linux), Board: "Arduino Due (Programming Port)"

Build options changed, rebuilding all
Using library DHT sensor library in folder: /home/pi/Arduino/libraries/DHT 
Using library Adafruit BMP085 Library in folder: /home/pi/Arduino/libraries/Adafruit_BMP085 
Using library Wire in folder: /home/pi/.arduino15/packages/arduino/hardware/sam/1.6.4/libraries/Wire 
Using library OneWire in folder: /home/pi/Arduino/libraries/OneWire (legacy)
Using library DallasTemperature in folder: /home/pi/Arduino/libraries/DallasTemperature (legacy)
Using library EmonLib in folder: /home/pi/Arduino/libraries/EmonLib (legacy)

/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -MMD -mcpu=cortex-m3 -DF_CPU=84000000L -DARDUINO=10605 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Due" -I/home/pi/.arduino15/packages/arduino/hardware/sam/1.6.4/system/libsam -I/home/pi/.arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/CMSIS/Include/ -I/home/pi/.arduino15/packages/arduino/hardware/sam/1.6.4/system/CMSIS/Device/ATMEL/ -I/home/pi/.arduino15/packages/arduino/hardware/sam/1.6.4/cores/arduino -I/home/pi/.arduino15/packages/arduino/hardware/sam/1.6.4/variants/arduino_due_x -I/home/pi/Arduino/libraries/DHT -I/home/pi/Arduino/libraries/Adafruit_BMP085 -I/home/pi/.arduino15/packages/arduino/hardware/sam/1.6.4/libraries/Wire -I/home/pi/Arduino/libraries/OneWire -I/home/pi/Arduino/libraries/DallasTemperature -I/home/pi/Arduino/libraries/EmonLib /tmp/build3513037533168774213.tmp/MeterNode.cpp -o /tmp/build3513037533168774213.tmp/MeterNode.cpp.o
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++: 1: /home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++: ELF??4DY: not found
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++: 2: /home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++: Syntax error: word unexpected (expecting ")")
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ returned 2
Error compiling.
<hr>
I tried compiling another sketch for a regular Arduino Leonardo & got the following output:

<hr>
Arduino: 1.6.5 (Linux), Board: "Arduino Leonardo"

Using library SoftwareSerial in folder: /home/pi/arduino-1.6.5/hardware/arduino/avr/libraries/SoftwareSerial

/home/pi/arduino-1.6.5/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 -DUSB_MANUFACTURER="Unknown" -DUSB_PRODUCT="Arduino Leonardo" -I/home/pi/arduino-1.6.5/hardware/arduino/avr/cores/arduino -I/home/pi/arduino-1.6.5/hardware/arduino/avr/variants/leonardo -I/home/pi/arduino-1.6.5/hardware/arduino/avr/libraries/SoftwareSerial /tmp/build3513037533168774213.tmp/MasterNode.cpp -o /tmp/build3513037533168774213.tmp/MasterNode.cpp.o

Cannot run program "/home/pi/arduino-1.6.5/hardware/tools/avr/bin/avr-g++" (in directory "."): error=2, No such file or directory
<hr>
Any help with getting this going would be greatly appreciated. Since the Pi & the Due are located in the electricity meter box, being able to remotely access the Pi & upload changes to the sketch to the Due would be extremely handy!

Thanks for your time!

transivit, 2015-10-10:
Has anyone been successful in putting together a succinct set of commands to activate the Arduino IDE 1.6.5 in Raspberry-Jessie yet?  I've stumbled through Crimier and NicoHoods' solutions to get past Arduino 1.0.1, but continue having major issues, with avrdude, non-connected serial ports, etc.

nielsnl68, 2015-10-17:
@CRImier hi, like @transivit  i like to know the same thing.

NicoHood, 2016-06-01:
@facchinm I think this can be closed now :)

closed 2016-06-06 by facchinm