arduino/Arduino #3335: "upgrading" IDE install process? - suggestion

VaclavSal, 2015-06-14:
I just installed an official 1.6.4. on my "new" PC and noted that it still deletes the older IDE version, including libraries of course.
With the recent "hourly builds" (really folks ?) and an official releases coming in short intervals it would be nice to really upgrade the install process and either delete / upgrade changed libraries only or give a user option to selectively do the same.

OK, I can copy the older libraries back to the new release , but how do I know if the library itself is the latest?

Chris--A, 2015-06-14:
Is it deleting the libraries in the user documents\Arduino\libraries folder?

VaclavSal, 2015-06-14:
Cannot say because I did not checked that folder before "upgrading".
At this point I need to verify where "contributed" libraries reside. The "Library Manager" does not give much info , despite "more info" button which only allows for install. 
I think the Library Manger still does not clearly identify the library in actual usage which has been a source of trouble for many users. But I have to admit it is definitely much need improvement.
I also need to test drive it more to get a real feel for its usage.

Chris--A, 2015-06-14:
The library manager installs them into the documents folder. A zip of the libs downloaded are stored in the Arduino15\staging folder.

I'm not concerned about the downloadable libraries. I'm worried about the many many libraries I have written, that aren't elsewhere. Sure I've got  a copy, but there will be many people that won't. 

It seems to use the older library (or first enumerated) which means the downloaded libs get second priority.

On  a side note. To get the latest builds (1.6.5) to use the correct AVR core, I had to delete the Arduino15 folder contents, as the AVR core has been moved back to the install folder, and it used the older 1.6.3 core.

VaclavSal, 2015-06-15:
To answer the original question - the libraries loaded onto 
Windoze documents directory do not get erased. 

So I assume that "contributed" libraries are in the Artduino15 staging folder as you mentioned and than "added" to this document folder to be used. Keeping in mind that not too long ago adding zipped library was a challenge it seems that all this new development work, however nice, is just building / patching on shaky foundation.
To the casual observer it should be notable that volunteer developers and an official Arduino folks do not seems to have common interests , especially in making the IDE more "user friendly".
I guess money is in selling  hardware.

ffissore, 2015-06-15:
Libraries (both from lib manager and manually installed) are all stored in SKETCHBOOK/libraries: such folder (as well as the sketchbook folder) does **not** get deleted.
The only folder that gets deleted is the IDE folder itself, and only when you use the windows installer and when you have a previously installed version (in essence, when you upgrade the IDE using the installer)
Closing this issue as invalid. If you have specific suggestions please try to be as brief as possible

closed 2015-06-15 by ffissore

VaclavSal, 2015-06-15:
I am puzzled by filsore response. 
The issue is invalid but added to release 1.6.5. 
So if it is invalid why bother?
Actually why did I bother to bring this up at all? 

As far as being brief and specific:

Upgrading IDE DOES not remove Windows directory and compiler Temp files are STILL not removed.

So - the main question - does anybody do QA before release?
Don't bother to respond, as far as I am concerned this is closed.